RobotLocomotion/drake #6647: examples/zmp and examples/QPInverseDynamicsForHumanoids are out of place

RussTedrake, 2017-07-24:
i think we need a top-level directory for legged_locomotion; these utilities should go in there (plus their tests / minimal examples), and the real examples (atlas, valkyrie, littledog, etc) should continue to use the from the examples directory.  

we also have the systems/robotInterfaces subdir, which also seems out of place, and might be another candidate for moving into legged_locomotion or equivalent dirs?

@siyuanfeng-tri -- what do you think?

siyuanfeng-tri, 2017-07-24:
I think that is the correct path, haven't got a real motivation to do it yet. Will do next.

jwnimmer-tri, 2019-06-20:
> ... `examples/zmp` and `examples/QPInverseDynamicsForHumanoids` are out of place ... we also have the `systems/robotInterfaces` subdir ...

`robotInterfaces` is in the attic.  `QPInverseDynamicsForHumanoids` is gone / in the attic.

We still have `examples/zmp` with `zmp_example`.  Do we want to move that to `systems/controllers` to hide it, or are we OK with it living where it currently does @siyuanfeng-tri @RussTedrake?

RussTedrake, 2019-06-21:
I think that the zmp_example is doing no harm there (though it looks like a good candidate for porting to python and zapping the callmatlab usecase).  fwiw -- a number of students in underactuated found and used this example just this past term.

some of the supporting files in that directory can be cleaned.  i'll PR that now.

jwnimmer-tri, 2019-06-24:
@RussTedrake you were looking at this.  Is there anything remaining to do?

RussTedrake, 2019-06-25:
i still think that repairing some of the locomotion examples and putting them in a legged_locomotion subdirectory is the right long-term solution.  But I think the current state is all that we will have for now.

closed 2019-06-25 by RussTedrake