RobotLocomotion/drake #6654: Add initial point cloud processing API to Drake

EricCousineau-TRI, 2017-07-25:
## Overview

- [ ] Implementation will be VTK-based, following suit with `RgbdCamera`, `director`, `LabelFusion`, etc. The Drake API will abstain from exposing VTK internals, like `RgbdCamera`.
    * If performance becomes an issue due to conversion, it may provide views into the VTK data, akin to `vtk.util.numpy_support`, by using `Eigen::Map<>`.
- [ ] I would still like to consider how to handle the API for point cloud types:
    * `drake::systems::sensors::Image` / PCL-style (template specializations based on enumerations)
        * PCL-style point cloud sub-indexing (for lighter-weight downsampling / selection, etc.) - may make it easier for direct association to pixels / pixel coordinates in an image.
    * VTK-style run-time specialization - most likely to go forward with this, if VTK is used under the hood.
- [ ] File formats for saving / loading point clouds directly will be focused on `*.vtp` files.

* To minimize external dependencies, PCL will not be included at the start. However, we can consider including Kitware's [`PCLPlugin`](https://github.com/Kitware/PCLPlugin) that is used in `director` as a direct bridge.
    * If we find PCL may provide better performance / features in the long run, we may consider switching API internals, but still maintain a bridge.
* Visualization will stay focused on the `director` / `drake-visualizer` pipeline (LCM conversion, etc.).

* The base API will first be developed independently of the System framework. Afterwards, System wrappers will be added.
    * The pipeline will be written in such a way that is amenable to the System framework's sparsity analysis (cleanly separating state, caching, etc). Most systems will be discrete-time for simplicity.
    * This design will not initially focus upon usage with `AutoDiff` / `symbolic` analysis. (The AutoDiff portion may be deferred to future work within Drake using TensorFlow or another machine learning framework).
        * As such, states will be `AbstractValue`s, and will not derive from `VectorBase<>`.

* To start, Drake will not do direct GPU acceleration, but instead delegate to third-party libraries.
    * CUDA support for `gcc` appears solid, but `clang`'s support for CUDA is still [under active development](https://llvm.org/docs/CompileCudaWithLLVM.html#id3), and therefore risky.
        * If we need GPU acceleration, then we may consider only supporting it under `gcc`.
        * Alternatives: OpenCL + SYCL

## Timeline
Basic perception algorithms will be implemented first, and then based on this, the point cloud API will be implemented.

\cc @kunimatsu-tri @gizatt

sherm1, 2017-07-25:
BTW, FCL also has point cloud support and is autodiffable. You may want to ask @avalenzu about that to see if it offers any advantages over VTK.

hongkai-dai, 2019-03-25:
@EricCousineau-TRI could you update the status of this issue? Thanks!

EricCousineau-TRI, 2019-03-25:
Similar to #6655; this is stale w.r.t. usage in Drake.

Copying from there:

> No longer relevant, as there was no concrete use case for this.
> There is some related work going on in Anzu; if/when it's appropriate to upstream, a separate issue should be used to track that work.
> 
> \cc @naveenoid @thduynguyen

closed 2019-03-25 by EricCousineau-TRI