RobotLocomotion/drake #6668: Data files in the installed binaries

stonier, 2017-07-26:
Potential list:

* *.urdf
* *.sdf
* *.obj
* *.stl
* ...

Any other files required by anyone?

@jwnimmer-tri where should these show up in the binary so that drake's resource path finder finds them?

jamiesnape, 2017-07-26:
Any chance that data files that are used by core and/or need installing could reside somewhere other than scattered throughout the examples?

stonier, 2017-07-26:
A potential attack vector might be to get it working for just one example (see https://github.com/RobotLocomotion/drake-shambhala/issues/11) and then shoot for the rest.

jwnimmer-tri, 2017-07-26:
> @jwnimmer-tri where should these show up in the binary so that drake's resource path finder finds them?

A file in the source tree at `$WORKSPACE/drake/examples/foo/bar.obj` should end up in
`$PREFIX/share/drake/drake/examples/foo/bar.obj`.

Note the double `drake`.  (The first is the repository name; the second is the initial element of the package name.)

Depending on how the downstream build system works, a developer may still need to set [DRAKE_RESOURCE_ROOT](http://drake.mit.edu/doxygen_cxx/namespacedrake.html#a329d1e7afba846a6ea6f8c2638fef7ec) environment variable to `$PREFIX/share/drake` to find the resources.

> Any chance that data files that are used by core and/or need installing could reside somewhere other than scattered throughout the examples?

I suspect that would be difficult.  I suspect that the list of "what should be installed" is approximately all model data that's not under a test folder, or perhaps only all _examples_ model data that's not under a test folder.

I think we're going to have to just whitelist a bunch of newly-written `:install` sub-targets under examples, perhaps some kind of `drake_install_filegroup` macro that helps DTRT with less boilerplate.  Or maybe there can be a `//drake/examples/:example_data` rollup that gets installed en masses, then its just a matter of listing out the models targets that it should collect.

jwnimmer-tri, 2017-07-26:
For now, let's only worry about data files from within `drake/manipulation/models` (and its subfolders) and `drake/examples` (and its subfolders).  Don't worry about other data files, e.g., `drake/multibody`.

fbudin69500, 2017-07-26:
`drake/examples/kuka_iiwa_arm` needs to be installed for **director**, so I'll start with `drake/examples` and then do the rest.

fbudin69500, 2017-07-26:
I created a PR (https://github.com/RobotLocomotion/drake/pull/6671). It is not ready to be merged, but it installs the data that director requires to run some tests.

stonier, 2017-08-09:
The files indicated by @jwnimmer-tri are in. At `share/drake/drake` seems obtuse, but might be necessitated by resource finding tools in drake. I'll check on that separately.

closed 2017-08-09 by stonier