RobotLocomotion/drake #6673: Do you have any function to calculate the ZMP of Atlas?

chang-liang, 2017-07-26:
Do you have any function to calculate the ZMP of Atlas?  Appreciate your help.

chang-liang, 2017-07-27:
In the file /drake-distro/drake/systems/robotInterfaces/@Biped/plotWalkingTraj.m, I found a ZMP calculation "zmpact(:,i)=com(1:2,i) - com(3,i)/9.81 * (J(1:2,:)*qdd + Jdotv(1:2));" However, the Jdotv is "NaN NaN NaN".

hongkai-dai, 2017-07-28:
What is the SHA of drake you are on? You can check that by typing the following command in the terminal
```
$ cd drake-distro
$ git log
```
could you tell me the number following the "commit"?

chang-liang, 2017-07-28:
commit 613022b9816eeef2b6fa8a398872ea2c3d32040c
Merge: b743b68 d286346
Author: Jeremy Nimmer <jeremy.nimmer@tri.global>
Date:   Thu Oct 6 17:04:55 2016 -0400

    Merge pull request #3707 from liangfok/feature/update_leaf_sys
    
    Temporary Fail Fast if Leaf System Non-Vector-Valued Port.

commit b743b68413274d5c06e72b1245c677fdc9057ff3
Merge: 585ddf6 24b252b
Author: David German <david.german@tri.global>
Date:   Thu Oct 6 16:26:48 2016 -0400

    Merge pull request #3706 from david-german-tri/modal_state
    
    Move DiagramContinuousState into its own file.

commit 585ddf669cb914be235df0b28c72f0f364988940
Merge: 7427d62 16ffe97
Author: Jamie Snape <jamiesnape@users.noreply.github.com>
Date:   Thu Oct 6 16:24:55 2016 -0400

:

chang-liang, 2017-07-28:
commit 613022b9816eeef2b6fa8a398872ea2c3d32040c
Merge: b743b68 d286346
Author: Jeremy Nimmer <jeremy.nimmer@tri.global>
Date:   Thu Oct 6 17:04:55 2016 -0400

    Merge pull request #3707 from liangfok/feature/update_leaf_sys
    
    Temporary Fail Fast if Leaf System Non-Vector-Valued Port.

commit b743b68413274d5c06e72b1245c677fdc9057ff3
Merge: 585ddf6 24b252b
Author: David German <david.german@tri.global>
Date:   Thu Oct 6 16:26:48 2016 -0400

    Merge pull request #3706 from david-german-tri/modal_state
    
    Move DiagramContinuousState into its own file.

commit 585ddf669cb914be235df0b28c72f0f364988940
Merge: 7427d62 16ffe97
Author: Jamie Snape <jamiesnape@users.noreply.github.com>
Date:   Thu Oct 6 16:24:55 2016 -0400

:...skipping...
commit 613022b9816eeef2b6fa8a398872ea2c3d32040c
Merge: b743b68 d286346
Author: Jeremy Nimmer <jeremy.nimmer@tri.global>
Date:   Thu Oct 6 17:04:55 2016 -0400

    Merge pull request #3707 from liangfok/feature/update_leaf_sys
    
    Temporary Fail Fast if Leaf System Non-Vector-Valued Port.

commit b743b68413274d5c06e72b1245c677fdc9057ff3
Merge: 585ddf6 24b252b
Author: David German <david.german@tri.global>
Date:   Thu Oct 6 16:26:48 2016 -0400

    Merge pull request #3706 from david-german-tri/modal_state
    
    Move DiagramContinuousState into its own file.

commit 585ddf669cb914be235df0b28c72f0f364988940
Merge: 7427d62 16ffe97
Author: Jamie Snape <jamiesnape@users.noreply.github.com>
Date:   Thu Oct 6 16:24:55 2016 -0400

    Merge pull request #3693 from soonho-tri/pr-editorconfig
:...skipping...
commit 613022b9816eeef2b6fa8a398872ea2c3d32040c
Merge: b743b68 d286346
Author: Jeremy Nimmer <jeremy.nimmer@tri.global>
Date:   Thu Oct 6 17:04:55 2016 -0400

    Merge pull request #3707 from liangfok/feature/update_leaf_sys
    
    Temporary Fail Fast if Leaf System Non-Vector-Valued Port.

commit b743b68413274d5c06e72b1245c677fdc9057ff3
Merge: 585ddf6 24b252b
Author: David German <david.german@tri.global>
Date:   Thu Oct 6 16:26:48 2016 -0400

    Merge pull request #3706 from david-german-tri/modal_state
    
    Move DiagramContinuousState into its own file.

commit 585ddf669cb914be235df0b28c72f0f364988940
Merge: 7427d62 16ffe97
Author: Jamie Snape <jamiesnape@users.noreply.github.com>
[... 466 lines cut ...]
    Merge pull request #3706 from david-german-tri/modal_state
    
    Move DiagramContinuousState into its own file.

commit 585ddf669cb914be235df0b28c72f0f364988940
Merge: 7427d62 16ffe97
Author: Jamie Snape <jamiesnape@users.noreply.github.com>
Date:   Thu Oct 6 16:24:55 2016 -0400

    Merge pull request #3693 from soonho-tri/pr-editorconfig
    
    Add EditorConfig file

commit d28634697d71a649f8ddcf039e597d130c03f7ed
Author: Chien-Liang Fok <liang.fok@tri.global>
Date:   Thu Oct 6 16:16:03 2016 -0400

    Added a fail-fast if a leaf system attempts to allocate an output port that'

commit 7427d62c0ed165f9ef7b4e1702670ddc0df43764
Merge: 3b6b8a0 bab71f5
Author: David German <david.german@tri.global>
Date:   Thu Oct 6 15:55:34 2016 -0400

:

chang-liang, 2017-07-28:
the result I posted in the comment here was not the same as what was showing on the terminal. you could check it in the file here.
[git log.txt](https://github.com/RobotLocomotion/drake/files/1182028/git.log.txt)

hongkai-dai, 2017-07-28:
The math to compute ZMP is what you have found
```
zmpact(:,i)=com(1:2,i) - com(3,i)/9.81 * (J(1:2,:)*qdd + Jdotv(1:2));
```
To compute this, you need to make sure that `J` and `Jdotv` are computed for the robot center of mass. You can do that by
```MATLAB
  kinsol = doKinematics(r, q, qd);

  [com(:,i),J]=getCOM(r,kinsol);
  Jdotv = centerOfMassJacobianDotTimesV(r,kinsol,0);

```

chang-liang, 2017-07-28:
Thank you! you are right. However, the value of `Jdotv` computed from `Jdotv = centerOfMassJacobianDotTimesV(r,kinsol,0);` is "NaN NaN NaN".

hongkai-dai, 2017-07-28:
Could you add a breakpoint at the beginning of `centerOfMassJacobianDotTimesV.m`, then check in which line it gives NaN?

chang-liang, 2017-07-28:
I found the reason.
 ` total_mass = getMass(obj, robotnum);` (in `/drake-distro/drake/systems/plants/@RigidBodyManipulator/centerOfMassJacobianDotTimesV.m`) 
`robotnum` is set to 0 and `getMass` returned 0 (total_mass), which caused "com_Jacobian_dot_times_v" is "NaN NaN NaN" (because of `  com_Jacobian_dot_times_v = Adot_times_v(4:6) / total_mass;`)

chang-liang, 2017-07-28:
so the code ` Jdotv = centerOfMassJacobianDotTimesV(r,kinsol,0);` in `drake-distro/drake/systems/robotInterfaces/@Biped/plotWalkingTraj.m` should be changed to ` Jdotv = centerOfMassJacobianDotTimesV(r,kinsol,1);` ?

hongkai-dai, 2017-07-28:
Great! Could you set `robotnum` to 1?

hongkai-dai, 2017-07-28:
> so the code Jdotv = centerOfMassJacobianDotTimesV(r,kinsol,0); in drake-distro/drake/systems/robotInterfaces/@Biped/plotWalkingTraj.m should be changed to Jdotv = centerOfMassJacobianDotTimesV(r,kinsol,1); ?

Exactly. 1 is the default value. I do not know why that code uses 0.

chang-liang, 2017-07-28:
Thank you, Hongkai! It works. But if I set` kinsol.mex` to 0, I got errors
`Reference to non-existent field 'T'.

Error in RigidBodyManipulator/inertiasInWorldFrame (line 25)
  body_to_world = kinsol.T{i};

Error in RigidBodyManipulator/centroidalMomentumMatrixDotTimesV>worldMomentumMatrixDotTimesV
(line 55)
  inertias_world = inertiasInWorldFrame(obj, kinsol);

Error in RigidBodyManipulator/centroidalMomentumMatrixDotTimesV (line 24)
    Adot_times_v = worldMomentumMatrixDotTimesV(obj, kinsol, robotnum);

Error in RigidBodyManipulator/centerOfMassJacobianDotTimesV (line 33)
    Adot_times_v = centroidalMomentumMatrixDotTimesV(obj, kinsol, robotnum);

Error in TimeSteppingRigidBodyManipulator/centerOfMassJacobianDotTimesV (line 1038)
      [varargout{:}] = centerOfMassJacobianDotTimesV(obj.manip,varargin{:});

Error in Biped/plotWalkingTraj (line 101)
  Jdotv = centerOfMassJacobianDotTimesV(r,kinsol,1);

Error in Biped/simulateWalking (line 34)
[com, rms_com] = obj.plotWalkingTraj(ytraj, walking_plan_data);

Error in Biped/runWalkingDemo (line 77)
[ytraj, com, rms_com] = obj.simulateWalking(walking_plan_data, walking_options); %took long
time on this line. Running on Drake for the first time when running on this line.

Error in runAtlasWalking (line 30)
r.runWalkingDemo(walking_options);`

hongkai-dai, 2017-07-28:
If you do not want to use the mex version, then you need to change the call to `doKinematics`, as
```MATLAB
kinsol = doKinematics(r, q, qd, struct('use_mex', false));
```

hongkai-dai, 2017-08-01:
I think we can close this issue?

chang-liang, 2017-08-01:
Ok, sure! Thank you and sorry for the late reply.

hongkai-dai, 2017-08-01:
No problem, thanks for your interest in Drake!

closed 2017-08-01 by hongkai-dai