RobotLocomotion/drake #6680: Install drake_visualizer

jwnimmer-tri, 2017-07-27:
It seems to me that `bazel run :install` should install a copy of `drake-visualizer` (maybe named `drake_visualizer`; whatever).   Drake's binary release will be much more usable if the built-in visualization tool works out of the box.

(Alternatively, our binary release could document how to download and install `drake-visualizer` from the Director binary releases, but that seems like it would not always like up with what our binary release of `pydrake` would be providing.  That seems worse.)

As a related benefit even for from-source users, installation might be a good way to solve the "Bazel runfiles sometimes vanish when I rebuild something else" headache.  A quick `bazel run //tools:install_drake_visualizer` would put the tool into known place that would not change until the user asked it to.  That may help people who like to keep the visualizer running for a long time while rebuilding other parts of the tree during development, and best matches the long-standing workflow.

stonier, 2017-07-28:
Beat me to posting this issue. 

+1 for getting this into the binary release while there is no conventional way of getting the drake-visualiser dependency in binary form.

No real opinion on the from-source install bazel rule. Perhaps that could be a separate issue?

jwnimmer-tri, 2017-07-28:
Is seems like the correct way to have `drake_visualizer` be a part of a binary release would be to make the `bazel run: install` rule install it, so I don't think its a separate issue.

stonier, 2017-08-06:
Agreed. Sounded like you were looking for a different alternative in your first post.

stonier, 2017-08-06:
The dependency chains here are so incredibly messy. Cannot just bring the director part in as is - there are eigen3, lcm and lcmtypes conflicts.

PR #6843 (2017-08-22): Download and install drake-visualizer built with static libraries
Resolves #6680. This also makes the Drake and RobotLocomotion LCM types available to drake-visualizer.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6843)
closed 2017-08-22 by jamiesnape