RobotLocomotion/drake #6685: Create Python bindings for find_resource.cc

fbudin69500, 2017-07-27:
Since `getDrakePath()`has been deprecated, it would be very useful to have access to the functions available in `find_resource.cc` [1] when looking for drake data.

[1] https://github.com/RobotLocomotion/drake/blob/master/drake/common/find_resource.cc

jwnimmer-tri, 2017-07-27:
For clarity -- the C++ `drake::GetDrakePath()` is deprecated, but the Python `pydrake.getDrakePath()` has always been a completely independent implementation.  The Python function still works just as well as it always did.

It's fine to also expose a `pybind.common.FindResourceOrThrow` if we want to, but I wanted to be clear that the deprecation mentioned in the OP is irrelevant.

fbudin69500, 2017-07-27:
@jamiesnape @jwnimmer-tri Should the function `pydrake.getDrakePath()` be updated for the drake install tree? Currently, the value returned by the function points to the `lib` subdirectory instead of the install directory (one directory lower). That being said, I am not sure where it points to in the bazel build directory, I haven't tested that.

jwnimmer-tri, 2017-07-27:
@fbudin69500 I have no idea.  The function has no documentation nor direct unit test, so I have no idea of what its semantics are supposed to be.  Looking at the pydrake unit tests that use it, it seems like its supposed to be a path such that appending "examples/foo/bar.urdf" locates a resource that originates in the drake source tree.

Thus, in the FHS-like install tree, where python code and model data are in separate folders, its might be too finicky to try to fix it.  Probably the right answer is to offer `pybind.common.FindResourceOrThrow` as this issue suggests, rather than trying to fix `pydrake.getDrakePath()`.

But that's just what I can tease out from the code.  Those more familiar with how `pydrake` is expected to be used can speak up if they know better.

fbudin69500, 2017-07-27:
@patmarion I encountered a problem with the example paths while updating Spartan to use drake/bazel, for directors' tests. What's your opinion on this matter?

patmarion, 2017-07-27:
`pydrake.getDrakePath()` should return the path to the top-level `drake/drake` dir, as can be seen in this usage:

https://github.com/RobotLocomotion/drake/blob/master/drake/bindings/python/pydrake/test/testRBTIK.py

Maybe it works in tests but does not work from the install tree?  Spartan has a macro to run drake and director tests, so if those tests run and pass then everything should be ok.

Callers of `pydrake.getDrakePath()` don't really care about the location of the drake source code, they care about the location of drake's standard resource files, like urdf and meshes.  So whatever works best to provide that is fine for uses in Director and Spartan.  It sounds like to me that `find_resource.cc` is a new and official way to provide these paths, so it seems like a good idea to add that to the drake python bindings.

jwnimmer-tri, 2017-07-27:
@fbudin69500 So while adding `FindResourceOrThrow` to `pydrake` is probably a good idea in general (and soon), if the purpose of `pydrake.getDrakePath()` is only to locate data resources such as urdf and meshes, then you could switch its implementation to call C++ `drake::common::GetDrakePath()` instead of the build-system-generated implementation, and I think everything would just work, including downstream projects without any changes.

fbudin69500, 2017-08-03:
@jwnimmer-tri : is the C++ `drake::common::GetDrakePath()` function accessible from Python?

jwnimmer-tri, 2017-08-03:
Not currently:
```
$ find drake/bindings -type f -print0 | xargs -0 grep -n -e GetDrakePath
Grep finished with no matches found at Thu Aug  3 11:32:04
```

Someone could (easily, I think?) add it into the `pybind11`.

fbudin69500, 2017-08-03:
The current implementation of `pydrake.getDrakePath()` does not require any additional information, and `drake::common::GetDrakePath()` is deprecated and generates the following warning:

```
drake/bindings/pybind11/pydrake_common.cc:43:33: warning: 'GetDrakePath' is deprecated: 
DRAKE DEPRECATED: Please use drake::FindResource() instead. [-Wdeprecated-declarations]
  m.def("GetDrakePath", &drake::GetDrakePath, 
                                ^
./drake/common/drake_path.h:12:13: note: 'GetDrakePath' has been explicitly marked deprecated here
std::string GetDrakePath();
            ^
1 warning generated.
```
To avoid this problem, I wrapped `FindResourceOrThrow` and replaced the usage of `pydrake.getDrakePath()` in director with `pydrake.common.FindResourceOrThrow(). This necessitates the configuration of the environment variable DRAKE_RESOURCE_ROOT but this is not an issue in director since there is a script (`setup_environment.sh`) to configure the environment.

jwnimmer-tri, 2017-08-03:
As for (2), the plan was always to update `drake::common::FindResource` (which is what `drake:common::GetDrakePath` uses under the hood) to know how to operate from an installed copy of Drake -- there is a `TODO` in the source file about it.  I hadn't done that yet because the installed tree layout of Drake wasn't ready yet at the time, but it should be possible to do it now.  Then, users would not have to set `DRAKE_RESOURCE_ROOT` (though they would be welcome to do so, if desired).

jwnimmer-tri, 2017-08-03:
As for (1), it would be reasonable to disable that warning at the call site within `bindings/pybind11`, if that was easier.

fbudin69500, 2017-08-03:
I created a PR to wrap `drake::ocmmon::FindResource`: https://github.com/RobotLocomotion/drake/pull/6766

As for finding drake resources without an environment variable, there is the issue of knowing where drake gets installed.

jwnimmer-tri, 2017-08-03:
> ... there is the issue of knowing where drake gets installed.

The pydrake code knows what filename its running from (it can ask Python).  We can step up from there to find `$INSTALL_ROOT` and then down into `share/drake`.  That should be a fixed transform versus pydrake's installed location, with no searching.

fbudin69500, 2017-08-03:
Although the fixed transform will be different in the source tree and the install tree.

jwnimmer-tri, 2017-08-03:
Right.  That would be just one of the search paths.  The `find_sentinel_dir()` would still have to be in play when running from the source tree.

fbudin69500, 2017-08-03:
I see two solutions to implement this:
1) improve `pydrake.getDrakePath()` to check both the source path and the install path, and see which one contains the sentinel. This is a matter of adding 2 lines to the already existing script, and the search paths are (can be) hardcoded.
2) use `pydrake::common::GetDrakePath()` and improve the search paths in `FindResource()` but this relies on the current working directory and the environment variable `DRAKE_RESOURCE_ROOT` so the Python script (which knows where it is) would need to update either. This does not seem optimal as the Python script basically needs to tell `FindResource()` where to look.

jwnimmer-tri, 2017-11-13:
From the posts above, I can't tell what remains to fix before this issue is closed.  Can someone clarify?

EricCousineau-TRI, 2018-12-08:
Fixed via #6766.

closed 2018-12-08 by EricCousineau-TRI