RobotLocomotion/drake #6691: Some System framework header files are too big

sherm1, 2017-07-27:
**Problem**: Currently the System framework headers provide all definitions inline (mostly). This provides maximum instantiation-type T flexibility, but makes it very difficult to work with these files. It is likely having a significant cost in compilation time also.

**Proposal**: use the `-inl.h` design pattern described [here](http://drake.mit.edu/cxx_inl.html) and used elsewhere in Drake. For an extended discussion see #3741. Ideally we could use the `extern template` approach described in [this comment](https://github.com/RobotLocomotion/drake/issues/3741#issuecomment-252684663), though rumor has it that extern template declarations may be fatal on OSX. That could be fixed with some `#ifdef`s though.

These are the worst offenders in systems/framework:

File        |    Size
---------|--------
system.h | 1650
leaf_system.h |  1240
diagram.h |  1525

I propose to split these three files first with the usual pre-instantiations; we can split more using the same pattern as they become unwieldy.

**Discussion topic**: @jwnimmer-tri expressed a preference (recently, not in #3741) for allowing system framework users to make novel instantiations including only the `.h` files, rather than requiring `-inl.h` inclusion as proposed in the [cxx-inl.h documentation](http://drake.mit.edu/cxx_inl.html). That would be done by including the `-inl.h` file at the bottom of the `.h` file as proposed in the comment linked to above. That works nicely if `extern template` is available. Otherwise I think it would be fine to require explicit `-inl.h` inclusion as we do elsewhere in Drake; I'm not sure system/framework needs to be treated differently than other Drake code. We can get rid of that requirement as soon as extern template is functional if it isn't already.

**Note**: apparently all Drake users are currently making do with the pre-instantiations we have (double, AutoDiffXd, Expression) since some required system/framework definitions exist only in .cc files (e.g. output_port.cc). To actually make it possible for people to make novel instantiations we would need to find those hidden definitions and move them to `-inl.h` files also.

Opinions solicited!

jwnimmer-tri, 2017-09-25:
If the goal is compile times, then we merely need to do some experiments, mostly related to how `extern template` works.  The major question is ...
- if we have both declarations and definitions coming in via `foo.h` (whether all in one header, or via a reverse-include of the `-inl.h`), and
- if we have an `extern template` in the same header,

... then is the compile time issue mitigated enough, or do we really have to exclude the definitions from the `.h`?  To my estimation, forcing users to list out a wide swath of `-inl.h` files is unlikely to scale, so we should try of avoid it if the trade-off is merely a modest difference in compile times.

However, it sounds like the major goal is to make it easier for Drake developers to edit these files.  That is a bit more subjective.

I guess my first inclination is to see if we can logically pare down the files, without resorting to file splitting.  For example, do we need every single method currently offered to be directly implemented on System or its helper subclasses?  For example, the allocation APIs are rarely mentioned or needed by users.  Perhaps they could be factored out into an interface, where only the (few) systems that care could provide a concrete implementation.  In that case, not only are the source files shorter, but the Doxygen APIs are also slimmed down, with advanced concepts moved into separate classes that can be cohesively documented on their own.

However, even if we do that, it would only get us so far.

Say we still have "too much" text in, e.g., `system.h`.  I could see an argument for consolidating the method APIs within a class declaration, exclusive of (most) implementations, so that they are easier to skim through.  We would do that by moving the bodies out of the class declaration, and into the namespace.  For the goal of Drake developers conveniently editing those files, whether we put the bodies at the bottom of the file, or into a separate `-inl.h` file, seems like a matter of personal preference, with pros and cons in both directions.  Thus, my inclination would be to stick with GSG and keep it all in one file.

So in short, pending compile-time performance and correctness experiments, my proposed goal would be:
- (1) add `extern template` decorations;
- (2) in too-big framework files, relocate method bodies from the the class declaration to the `systems` namespace after the class has been declared;
- (3) start to dis-prefer and deprecate the `-inl.h` pattern.

sherm1, 2017-09-25:
Assuming we intend to fix the list of possible instantiations to a finite set, for which #7093 is good progress, putting method definitions in the .cc file is a good option.

jwnimmer-tri, 2020-02-10:
Here's an update on my current posture: we should drop framework for user-supplied scalars, and only support what's built into Drake (currently double, AutoDiffXd, and Expression).  So yes, we should move method bodies to the cc file when we don't need them to be available for inlining.

hongkai-dai, 2020-02-10:
I would like to support nested autodiff scalar `AutoDiffX<AutoDiffXd>` so that we can compute the Hessian. We have github issue #12019 and #4267 that explain the usage of Hessian in optimization. With the Hessian of the constraint/cost, we can compute how a change on the constraint/cost parameter would affect the optimal solution.

jwnimmer-tri, 2020-02-10:
Mostly that would just be adding it to the list at https://github.com/RobotLocomotion/drake/blob/master/common/default_scalars.h.

sherm1, 2020-05-11:
Much code has moved .cc. The headers are still big but now mostly documentation. Feel free to open a new issue if you spot particular framework files that need cleanup.

closed 2020-05-11 by sherm1