RobotLocomotion/drake #6698: PR process for external contributions

stonier, 2017-07-28:
Right now the process has some bumps when an external contributor creates a pull request. I'd like to see this become smoother and for us to start practicing it sooner rather than later. Current problems:

* The PR discussion thread gets spammed with "Can one of the admins verify this patch?" 
* Our documentation does not reflect the process for external contributions
* Could use an auto-inserted paragraph in the PR post pointing to the reviewing process documentation

Should check for other hitches that prevent them from fully interacting in the review process also. 

Note also, that external contributors do not have the permissions to do the usual "+@xyz for feature review please" which need not change. Drake developers should be aware that they are responsible for assigning a reviewer where none is present. I do not think any special process is needed here, that is the conventional workflow for open source projects but we may at some point want to delegate certain drake devs to be responsible for triaging  contributions when they come in more rapidly.

Any other thoughts?

jwnimmer-tri, 2017-07-28:
FYI the first bullet point is already filed as #3537.

jwnimmer-tri, 2018-06-09:
Possibly worth revisiting the second and third points again now (relates #8851).

RussTedrake, 2018-08-31:
the github PRs now have a template which helps people through the process.  is that a sufficient resolution?  
@jwnimmer-tri -- back at you.  Might be closable?

closed 2018-10-28 by jwnimmer-tri