RobotLocomotion/drake #6702: Review process to keep monolane and multilane in sync

jwnimmer-tri, 2017-07-29:
Per discussion in #6639, we are going to use code review to keep the in-tree fork of `monolane` and `multilane` code in sync.  We should make sure that platform reviewers know about this, and also ideally teach reviewable that we need a specific emoji on any PRs that touch `monolane`.

/CC @agalbachicar @maddog-tri

jwnimmer-tri, 2017-08-01:
The reviewable interlock is done and tested on #6724.  PRs now need a :car: emoji posted by myself or the two folks tagged in the above anytime they touch `monolane`.  The responsibility of our review is to make sure the `monolane` edit is also merged into `multilane`.

closed 2017-08-01 by jwnimmer-tri

maddog-tri, 2017-08-01:
:+1:  Thanks for setting up the machinery.