RobotLocomotion/drake #6707: Testing call_matlab_client

RussTedrake, 2017-07-29:
The newer version of CallMatlab writes to a file (via protobuf) during execution of the c++ program; then you can run matlab at any time on that file to visualize the results.  I would like to put test coverage not only on the call_matlab_client implementation, but also the matlab code inside the c++.  What's the right way to do this in our current testing framework?

I think i'd basically like to do a (cmake) add_matlab_test that calls the c++ target first, then calls the matlab client and makes sure it terminates without errors.  I also want to plan ahead for the near future when the c++ is only available via bazel, even though the matlab client is only available/testable via cmake.

jamiesnape, 2017-07-31:
Can you share a MATLAB script that would (roughly) implement the test?

jamiesnape, 2017-07-31:
Aside: This TODO needs to be implemented for CI: https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/call_matlab_client.cc#L33-L34
We cannot guarantee that a location in `/tmp` will be writeable and if it is, whether it will be cleaned up between successive builds.

RussTedrake, 2017-07-31:
understood that we need to implement the file name option.  Then call_matlab_client would be the (only) command that matlab runs; if it completes without errors then the test passes.

jwnimmer-tri, 2017-07-31:
I haven't entirely studied where all the pieces come from, but if for some reason running both programs (client and server) in the same build is difficult, then we could also commit the intermediate file, and one one side test that the C++ program emits exactly those bytes, and that those bytes fed into MATLAB pass the test.  And if the bytes ever need to change (unlikely), we just commit whatever the C++ is emitting at the time.

RussTedrake, 2017-08-09:
fyi -- i have a WIP for this now.  and I've confirmed that an updated version of the symlink instructions in #960 are still mandatory for matlab 2017a on ubuntu.  @pangtao22 -- perhaps something similar is happening to you on mac?

EricCousineau-TRI, 2019-08-08:
No longer relevant given resolution in #11753. Closing.

PR #11909 (2019-08-08): Remove MATLAB-specific code from call_matlab
Relates #11909. Closes #6707.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11909)
closed 2019-08-08 by EricCousineau-TRI