RobotLocomotion/drake #6727: symbolic::Expression::Display should never approximate doubles

jwnimmer-tri, 2017-08-01:
In `ExpressionConstant::Display` we are careful to use `max_digits10` for printing, but in places like `-Add` and `-Mul` for the constant, we fail to do so.  (I think those are the only two we've missed.)  All `double`s printed by `Display` should use full precision.

jwnimmer-tri, 2017-08-01:
It would also be super cute if I ever did `T{M_PI}` (or similarly for `T{M_E}`, maybe even some of the `M_SQRT` stuff) where the constant was bit-exact with a fundamental constant that it would keep that term separate during constant-folding and print it symbolically.  So for example `T{M_PI} / 100.0` would print as `PI / 100.0` not `0.0314...`.

sherm1, 2017-08-01:
π would be an approximation of M_PI since the latter is a particular finite double precision number, different from π. Better I think would be to have symbolic constants from which numerical approximations could be obtained when needed.

soonho-tri, 2017-08-02:
@jwnimmer-tri , I think we can close this one by #6735? Please feel free to open another issue if you want to have symbolic constants for `PI` or `E`.

closed 2017-08-02 by jwnimmer-tri