RobotLocomotion/drake #6730: Printing Systems' symbolic form

jwnimmer-tri, 2017-08-01:
Unit testing the Systems that support symbolic form is a bit annoying.  Just asking about input-output sparsity gets you some of the way there, but really we should be examining the expressions that come out, either their exact form, or their detailed variable-to-variable sparsity including e.g. state and parameters (not just inputs), or something.

I suspect that a fair way to do this would be to have a method that operates on `System` and returns the string form of its symbolic equations.  It could be a method on `System` itself, or a free function.

For example:
```c++
  EXPECT_EQ(
      systems::PrintSymbolicEqations(dut),
      "Symbolic equations:\n"
      "y0_0 = ((0.63662 * floor(((100 * u0_0) / 0.63661977236758138))) / 100)\n"
      "y0_1 = ((0.63662 * floor(((50 * u0_1) / 0.63661977236758138))) / 50)\n");
```

Similarly, time derivatives or difference equations could be printed, or perhaps even validity constraints or guard functions.

I am curious if anyone else wants such a printer function, or there are other related purposes that could also be folded into this.  (Can `GraphViz` format do hovertext?  Maybe the graphviz output should have hovertext presenting the `Expression` for each output.)

RussTedrake, 2017-08-09:
I think the printing functionality is extremely useful, and worked right out of the box when I tried it here:
https://github.com/RobotLocomotion/drake/blob/master/drake/examples/pendulum/print_symbolic_dynamics.cc

But it seems like a lousy way to check the correctness of the symbolic form.  I would recommend instead that we actually evaluate the symbolic expression at some values and check the doubles that come out, e.g. using
https://github.com/RobotLocomotion/drake/blob/8dec3da2f9565f56417f8a288c575bd0301e8840/drake/common/symbolic_environment.h#L24

jwnimmer-tri, 2017-09-06:
Yeah, I no longer think printing is going to be useful as a unit test.

closed 2017-09-06 by jwnimmer-tri