RobotLocomotion/drake #6731: We have both drake::SpatialForce and drake::multibody::SpatialForce defined

sherm1, 2017-08-01:
drake::SpatialForce is defined in common/eigen_types.h basically as a typedef for a Vector6, and is only lightly used within RigidBodyPlant and Valkyrie. OTOH, drake::multibody::SpatialForce is a force-space SpatialVector providing essential functionality in the new MultibodyTree code. 

The names conflict in typical programs that include namespace drake and drake::multibody (discovered by @mitiguy while making multibody tests). We should get rid of the typedef asap. Since the code it is in will soon be deprecated, it would be fine to give it a somewhat awkward name like RBSpatialForce. Many other solutions are possible.

amcastro-tri, 2017-08-01:
Thank you for filing this issue, I had forgotten about this typedef.

amcastro-tri, 2017-08-01:
This is the typedef in `eigen_types.h`:
`template <typename Scalar> using SpatialForce = Eigen::Matrix<Scalar, 6, 1>;`

Within the same file we have a more appropriate typedef for this, `Vector6<T>`, in consistency with all other typedefs in the same file. If you all agree I will go ahead and replace `drake::SpatialForce<T>` with `drake::Vector6<T>` to avoid any possible collisions for now until this code is deprecated. RBP and Valkyrie code could actually be easily update to use `drake::multibody::SpatialForce<T>`.

sherm1, 2017-08-01:
Per f2f we need to look more closely at @mitiguy's test to see why he's seeing a conflict but @amcastro-tri doesn't.

sherm1, 2019-04-19:
Recently rediscovered. Relevant [Slack discussion](https://drakedevelopers.slack.com/archives/C2ETTRXT7/p1555512929000900).

amcastro-tri, 2019-04-19:
12 uses of `SpatialForce<double>` in the attic. We could replace with `Vector6<double>` and remove `drake::SpatialForce` from `eigen_types.h`?

PR #11806 (2019-07-17): Deprecate `drake::SpatialForce`
Resolves #6731

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11806)
closed 2019-07-17 by EricCousineau-TRI