RobotLocomotion/drake #6732: need code for solving lyapunov equations in c++

RussTedrake, 2017-08-01:
analogous to our algebraic riccati equation codes, but for the lyapunov equations
(e.g. https://www.mathworks.com/help/control/ref/lyap.html)

we need this to bring over our region of attraction code.
@psiorx , @FischerGundlach , ... ?  any takers?

FischerGundlach, 2017-08-01:
@RussTedrake Please assign me!

Further, I found this code: http://slicot.org/objects/software/shared/doc/SB03MD.html
The matlab version is based on this as well. 

Low priority question: 

What do we want to pass as input? We consider the system 
```A^T*X+X*A+Q=0```
So we definitely want `A` and `Q`, but we could additionally pass `S` and `U`, where 
```A=U*S*U^T```, is the Schur factorization. We need this factorization while computing `X`. 

The question is: does someone (a different caller) needs the factorization as well, such that it is worth it to keep it outside rather pass it in? Or should we provide both APIs?

It would be similar to `math::ContinuousAlgebraicRiccatiEquation` which has 2 APIs.

RussTedrake, 2017-08-01:
thanks @FischerGundlach .  i only need A and Q as inputs.  Let's stick with the simpler API unless/until we have a use case for the schur factorization version.

FischerGundlach, 2017-08-09:
So, there is code for the ContinousLyapunovEquation in here: https://github.com/RobotLocomotion/drake/pull/6795

The version I implemented supports Lyapunov equations with `A` and `Q` as real values matrices. Is there the need to handle complex valued systems? The code for complex systems would be similar, but somewhere along the way a different factorization is used.

I added some testing, but there might be more necessary/ wished. For a system in 3 states, the equation is solved in less than a millisecond, compared to Matlabs `lyap` which take roughly 12 milliseconds. Both versions should scale similarly with increasing state amount, as both are based on the same algorithm. 

I will add the utility for the discrete version soon. The code is similar, but I think different enough to place it in a complete own file. (The steps are the same, but the actual equations that need to be solved are different.)

RussTedrake, 2017-08-10:
Excellent.  I don't think we need to complex A/Q support at this time.

FischerGundlach, 2017-08-10:
Code for discrete case is (provisionally) in the same PR and pushed!

RussTedrake, 2019-03-08:
@weiqiao -- i just found this.

FischerGundlach, 2019-03-08:
@weiqiao 

I was pinged and saw that the [PR](https://github.com/RobotLocomotion/drake/pull/6795)  still needed some changes. 

I am currently on holidays, but ask me if there is something I can help on with + if the code/ parts of it make it into the master I am very happy.  If I remember, both continouse and discrete were tested.

weiqiao, 2019-03-09:
@FischerGundlach 
I think you code is good enough. If I must make some changes, I would
- replace the recursive call by an iterative function to save memory for large inputs
- write the function in template format

FischerGundlach, 2019-03-09:
When would you need it running by?

weiqiao, 2019-03-09:
@FischerGundlach No worries. I'll take care of it.

RussTedrake, 2019-03-30:
it's in.  hoorah!

closed 2019-03-30 by RussTedrake