RobotLocomotion/drake #6746: Installed SDFormat library missing dependencies

stonier, 2017-08-02:
```
$ ldd /opt/drake/lib/libsdformat.so
linux-vdso.so.1 =>  (0x00007fff73158000)
libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f136ba5d000)
libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f136b754000)
libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f136b53d000)
libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f136b173000)
/lib64/ld-linux-x86-64.so.2 (0x000055e3e620d000)
```

Seems to be missing both `libignition_math.so` and `libtinyxml.so`.

These problems are turning up when trying to link the example in https://github.com/RobotLocomotion/drake-shambhala/pull/22 -> undefined references to `TiXml***`. Since the example does not use these directly, they should be transitively picked up by a dependency on `sdformat`.

It looks like tinyxml is missed in [sdformat-create-cps.py](https://github.com/RobotLocomotion/drake/blob/master/tools/sdformat-create-cps.py) and both tinyxml and ignition math missed in the component dependencies. This does not explain why the installed `libsdformat.so` doesn't get linked against these libraries though.

jamiesnape, 2017-08-02:
The linking issue is a long-standing Bazel problem. We can certain update the CPS to ameliorate it, though.

jamiesnape, 2017-08-02:
https://github.com/bazelbuild/bazel/issues/492

PR #6817 (2017-08-14): Fix requirements/link flags in Ignition RNDF and SDFormat CPS files
Fixes #6746. Note that this is not going to change the output of `ldd /opt/drake/lib/libsdformat.so`. That is a Bazel problem that they will have to fix.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6817)
closed 2017-08-14 by jwnimmer-tri