RobotLocomotion/drake #6754: Multilane Segment Geometry

agalbachicar, 2017-08-02:
As part of #4934, the aim of this issue is to propose the creation of a `ReferenceGeometry` class that holds  the information of a reference curve composition along the `Segment`'s surface. This `ReferenceGeometry` entity will hold the following information:

- A cubic polynomial to model the _superelevation_.
- A cubic polynomial to model the _elevation_.
- Virtual methods to implement position and heading interpolation (and its derivatives) with respect a _p_ variable. This methods will model the reference curve over a plane _z=0_.

Specifying this base `ReferenceGeometry` class will model both line and arc geometries as they are already done in `monolane::LineLane` and in `monolane::ArcLane` but at the Segment level. `multilane::Segment` will hold an implementation of these `ReferenceGeometry` and take all the `monolane::Lane` geometric and generic methods and let the multilane::Lane node about:

- `do_length`
- `DoToGeoPosition`
- `DoToLanePosition`
- `DoGetOrientation`
- `DoEvalMotionDerivatives`

These are the changes in the current Multilane classes that should be done:

- `ReferenceGeometry` and its specifications should be implemented.
- `multilane::Segment`'s constructor should receive the parameters to create the `ReferenceGeometry`.
- A `multilane::Segment` will provide an API to create a `multilane::Lane`. Through the `Lane::segment()` call, it will have access to all the necessary methods to comply with Maliput's API.
- Move the geometric methods from `multilane::Lane` to `multilane::Segment`. Remove the `multilane::LineLane` and `mulitlane::ArcLane` classes, as its implementation will be handled at the `Segment` level.

agalbachicar, 2017-08-02:
@maddog-tri, which is your opinion?

maddog-tri, 2017-08-03:
Is the idea to split Lane into Lane + ReferenceGeometry first, or to promote *Lane logic into *Segment logic first?  (Either is fine by me --- whatever is the easiest to PR while maintaining unit-test coverage along the way.)

> `multilane::Segment`'s constructor should receive the parameters to create the `ReferenceGeometry.`

I think it'd be cleaner if the `Segment` constructor simply received an instance of `ReferenceGeometry` (e.g., via `unique_ptr`); it shouldn't have to know anything about how the various `ReferenceGeometry` subclasses are constructed.

> A `multilane::Segment` will provide an API to create a `multilane::Lane.`

Conversely, given the regular structure of the lanes described in #4934, the `Segment` constructor should probably take the defining parameters (# of lanes, spacing, offset to reference, ...) and construct the appropriate bundle of `Lanes` by itself.  And, for the very first pass through this, there will only be a single Lane anyway (i.e., refactoring while maintaining exactly the same feature-set as the `monolane` code).

agalbachicar, 2018-07-03:
Closing this issue as all the code is already merged in master.

closed 2018-07-03 by agalbachicar