RobotLocomotion/drake #6767: Drake lint produce line error output

SeanCurtis-TRI, 2017-08-03:
When drake lint detects out-of-order header includes, it would be good if it could enumerate them in the output. Specifically, if it had a format that was:

[Full File Path]:[line_no] 

Those two pieces of data can be followed by arbitrary messages. This will enable clickable links in CLion.  It is important that they be absolute paths; CLion doesn't handle relative paths at all.  :(

jwnimmer-tri, 2017-09-01:
@SeanCurtis-TRI Do you happen to know if CLion understand the standard "Entering directory" build output header, e.g.,
```console
blaze: Entering directory `/home/jwnimmer/.cache/bazel/_bazel_jwnimmer/8d1e0fabf4d48d295f3b4d143e04c94c/execroot/drake/'
```
or similar?  If a line like that appears first, can we then use relative paths?

SeanCurtis-TRI, 2017-09-01:
To the best of my knowledge, if I read this right, CLion doesn't know jack.

It has a line-by-line regular expression (and an incredibly limited expression at that). All information it needs must be on a single line. :(

jwnimmer-tri, 2017-09-01:
Understood.

So in http://drake.mit.edu/clion.html#correct-include-ordering, we are invoking `bazel run` for some linters.  It seems like we could make the absolute-path reporting opt-in, so that you would do `run //drake/tools:clang-format-includes -- --show-absolute-paths $FilePath$`.

Actually, is `$FilePath$` already absolute in CLion?  If so, I could just have the tool echo back what it was given, since in the build lint, the paths will be relative already.

SeanCurtis-TRI, 2017-09-01:
`$FilePath$` is *currently* absolute path. If it ceases to be absolute path in future versions (unlikely) we can amend instructions accordingly.

PR #6978 (2017-09-06): Report filename and line in drakelint error messages
Fixes #6767.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6978)
closed 2017-09-06 by jwnimmer-tri