RobotLocomotion/drake #6770: Modernize cpplint_wrapper.py

jwnimmer-tri, 2017-08-03:
Right now, `cpplint_wrapper.py` calls out to the `git submodule` flavor of `google_styleguide`.  We want to stop using submodules, so at a minimum it needs to switch over Bazel's download of the tool.  Possibly the script should disappear entirely, now that "lint all the files" is part of the build system.  If so, we'll need to update `drake/doc/clion.rst` with the new recipe.

SeanCurtis-TRI, 2017-08-03:
Just to echo a point previously discussed in a slack conversation.  The "lint all the files" of the current state of affairs produces feedback of the form `[relative file name]:[line number]`. In its infinite wisdom, CLion can only turn *absolute* paths into clickable links. So, if/when we pull the trigger on this, we either need to make paths absolute (at least configurable to such) or update the clion documentation to eliminate the external tool that recommends this tool.

jwnimmer-tri, 2017-09-01:
See also #6767 re: paths.

PR #6979 (2017-09-06): Update cpplint_wrapper for Bazel
Closes #6770.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6979)
closed 2017-09-06 by jwnimmer-tri