RobotLocomotion/drake #6779: Automotive demo instructions not working out of the box

cathywu, 2017-08-04:
Following the setup instructions, I get the following error when trying to run the automotive demo. I'm wondering if the instructions are out of date. I am running on Ubuntu 16.04.2 LTS

```
~/drake-distro$ bazel run drake/automotive:demo  -- --num_simple_car=1 --driving_command_gui_names=0 --num_trajectory_car=1
...............
INFO: Found 1 target...
INFO: From Executing genrule //drake/automotive:speed_bump_genrule:
[2017-08-04 12:21:57.852] [console] [info] Loading road geometry.
[2017-08-04 12:21:57.853] [console] [info] Generating OBJ.
Target //drake/automotive:demo up-to-date:
  bazel-bin/drake/automotive/demo
INFO: Elapsed time: 380.213s, Critical Path: 142.21s

INFO: Running command line: bazel-bin/drake/automotive/demo '--num_simple_car=1' '--driving_command_gui_names=0' '--num_trajectory_car=1'
*** LCM logs will be in /home/cathy/.cache/bazel/_bazel_cathy/a4b41a9f294fa1e9b737057fac57368d/execroot/drake/bazel-out/clang-3.9-linux-opt/bin/drake/automotive/demo.runfiles/drake
Traceback (most recent call last):
  File "/home/cathy/.cache/bazel/_bazel_cathy/a4b41a9f294fa1e9b737057fac57368d/execroot/drake/bazel-out/clang-3.9-linux-opt/bin/drake/automotive/demo.runfiles/drake/drake/automotive/automotive_demo.py", line 254, in <module>
    main()
  File "/home/cathy/.cache/bazel/_bazel_cathy/a4b41a9f294fa1e9b737057fac57368d/execroot/drake/bazel-out/clang-3.9-linux-opt/bin/drake/automotive/demo.runfiles/drake/drake/automotive/automotive_demo.py", line 235, in main
    wait_for_lcm_message_on_channel('DRAKE_VIEWER_STATUS')
  File "/home/cathy/.cache/bazel/_bazel_cathy/a4b41a9f294fa1e9b737057fac57368d/execroot/drake/bazel-out/clang-3.9-linux-opt/bin/drake/automotive/demo.runfiles/drake/drake/automotive/automotive_demo.py", line 176, in wait_for_lcm_message_on_channel
    "Timeout waiting for channel %s" % channel)
RuntimeError: Timeout waiting for channel DRAKE_VIEWER_STATUS
ERROR: Non-zero return code '1' from command: Process exited with status 1.
```

rdeits, 2017-08-06:
Hi Cathy! I'm not familiar at all with the automotive stack (and I'm barely familiar with the bazel build), but I think the error message you're seeing is the result of the `drake-visualizer` executable not starting up properly. To get more information about what's wrong, you'll want to run that binary (`drake-visualizer`) itself, so we can see how it's failing. Alas, I don't actually know how to invoke the visualizer from the bazel build, but surely someone else here does 😄

soonho-tri, 2017-08-07:
Hi @cathywu , could you try `bazel run //tools:drake_visualizer`?

soonho-tri, 2017-08-11:
Hi @cathywu , do you still have the problem? When you have time, could you test `bazel run //tools:drake_visualizer` and share the output with us?

cathywu, 2017-08-16:
Thank you for your help, I no longer need help with this!

soonho-tri, 2017-08-16:
Great! I'm closing this one.

closed 2017-08-16 by soonho-tri