RobotLocomotion/drake #6784: drake_visualizer is not working on mac

soonho-tri, 2017-08-07:
Repro steps:
 - `bazel clean`
 - `brew update && brew upgrade && brew cleanup -a`
 - `bazel run //tools:drake_visualizer`

Error message:
```
INFO: Running command line: bazel-bin/tools/drake_visualizer
dyld: Library not loaded: /usr/local/opt/glew/lib/libGLEW.2.0.0.dylib
  Referenced from: /private/var/tmp/_bazel_soonhok/51e1ef8bb59598000e1968965dd33f03/execroot/drake/bazel-out/osx-opt/bin/tools/drake_visualizer.runfiles/drake/external/director/bin/drake-visualizer
  Reason: image not found
```

FYI, `glew` was bumped to `2.1.0` recently. I guess that we have the problem because the bottled `robotlocomotion/director/vtk@8.0` is pointing to `glew-2.0.0`?

jamiesnape, 2017-08-08:
`brew reinstall vtk@8.0`.

soonho-tri, 2017-08-08:
I have:
```
INFO: Running command line: bazel-bin/tools/drake_visualizer
dyld: Library not loaded: /usr/local/lib/libjpeg.8.dylib
  Referenced from: /private/var/tmp/_bazel_soonho.kong/c11e5985260ea661cd5a496c449c8e9c/execroot/drake/bazel-out/osx-opt/bin/tools/drake_visualizer.runfiles/drake/external/director/bin/drake-visualizer
```
There is only `/usr/local/lib/libjpeg.9.dylib`.

jamiesnape, 2017-08-08:
Ok, the drake-visualizer package must need rebuilding too. I will look into it.

PR #6816 (2017-08-10): Fix drake-visualizer on Mac
Fixes #6784.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6816)
closed 2017-08-10 by jamiesnape