RobotLocomotion/drake #6800: MATLAB hangs without any error message after executing call_matlab_client

pangtao22, 2017-08-09:
Here's a screenshot of the symptom: 
[image]

I tried adding a mexPrintf a the very beginning of the mexFunction (https://github.com/pangtao22/drake/blob/debug_call_matlab_client/drake/matlab/util/call_matlab_client.cc#L33), but even that is not printed.

pangtao22, 2017-08-09:
@RussTedrake

RussTedrake, 2017-08-09:
please see the discussion i put this morning into #6707

RussTedrake, 2017-08-10:
It took me a day to install matlab on mac again, but i've just verified that call_matlab_client worked as expected for me... admittedly with R2017a.  

i don't see any call to `addpath_drake` in your command history, and i don't see the `call_matlab_client` in your current directory view.  did you save the drake path's in your matlab path directly?  (if so, that's not recommended... we might change `addpath_drake` at any time).  let's make absolutely sure that you are calling the right version of the mex file.

pangtao22, 2017-08-10:
I upgraded to R2017a and eveything works as expected! 

And I am pretty sure that I've been calling the right mex file:
```
>> which call_matlab_client
/Users/dideng/drake-distro/drake/matlab/util/call_matlab_client.mexmaci64
```
BTW, it took me less than an hour to install the new MATLAB.

RussTedrake, 2017-08-10:
Ok.  Good.  But that actually means we still need to figure out why it didn't work in 2016a.

hongkai-dai, 2017-08-10:
I used call_matlab_client on MATLAB 2016a, Linux 16.04, and it works great.

RussTedrake, 2017-08-11:
just confirmed that it hangs for me on mac 2016a.  will investigate.

RussTedrake, 2017-08-11:
I've verified that it hangs even if the code in the mex file is commented out (it simply returns).  
If I simply remove drakeCommonProto from the target_link_libraries, then it runs successfully again.  So I believe there is a library dependency clash happening here, which is not happening in 2017a.    
I've updated the issue title to reflect this finding.

RussTedrake, 2017-08-11:
further confirmed that the difference in libraries are
```
	@rpath/libdrakeCommonProto.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libprotobuf.dylib (compatibility version 0.0.0, current version 0.0.0)
```
but that linking directly against PROTO_LIBRARIES is ok, so the symbol clash much be in 
```
	@rpath/libdrakeCommonProto.dylib (compatibility version 0.0.0, current version 0.0.0)
```

interestingly, severing the drakeCommon dependency from drakeCommonProto did dramatically simplify the linker output
```
Russs-MacBook-Pro% otool -L call_matlab_client.mexmaci64
call_matlab_client.mexmaci64:
	@rpath/call_matlab_client.mexmaci64 (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libdrakeCommonProto.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/drakeMexUtil.mexmaci64 (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libprotobuf.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libmx.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libmex.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.5.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.60.2)
```
but the problem persists.

Further identified that the problem goes away if I remove `PROTO_SRCS` (and therefore necessarily `call_matlab.cc`) from `drakeCommonProto`.

RussTedrake, 2017-08-11:
I'm going to stop pursuing this for now.  Steps to reproduce my workflow:
```
cd drake-distro/drake/matlab
/Applications/MATLAB_R2016a.app/bin/matlab -nodesktop -nosplash -r "addpath_drake; call_matlab_client; exit"
```
this is the test that either exits (on success) or hangs (on failure).  by default, on master, it will hang.

```
cd drake-distro/build/drake
cmake . -DMatlab_ROOT_DIR="/Applications/MATLAB_R2016a.app"; make call_matlab_client
```
will rebuild (selectively) after any changes, and ensure that we're linking against the correct versions of matlab.

to make minimal changes that result in a successful execution of the test above:
- replace the entire contents of the `mexFunction` method in `call_matlab_client.cc` with a simple `return` statement.  the test above will still fail.  
- edit `drake-distro/drake/common/proto/CMakeLists.txt` and to comment out `call_matlab.cc` from `sources`.  the test above will still fail.  
- edit `drake-distro/drake/common/proto/CMakeLists.txt` and to comment out `${PROTO_SRCS}` from `sources`.  the test above will now pass.

commands like `otool -L call_matlab_client.mexmaci64` are very helpful.  But I don't see any differences (even recursing down) with the minimal succeed/fail above.  i expected to find conflicts with the contents of `/Applications/MATLAB_R2016a.app/sys/os/maci64`, but have not so far.

I also tried changing the name of the messages in `matlab_rpc.proto`, just on a hail mary.  as expected, that did not help.

RussTedrake, 2017-08-11:
For completeness (on mac sierra)
MATLAB R2017a works fine (does not exhibit this problem).
MATLAB R2016b still has this problem.  
MATLAB R2015b claims "no support for mac sierra", so I didn't try it.

RussTedrake, 2017-09-07:
i think we've documented the current state of the world.  I'm going to close this as "won't fix".

closed 2017-09-07 by RussTedrake