RobotLocomotion/drake #6803: RigidBodyTree's removeCollisionGroupsIf method produces segmentation faults supposedly because of bullet

TristanThrush, 2017-08-09:
The way that I'm currently trying to remove collisions in c++ is by using RigidBodyTree's removeCollisionGroupsIf. I pass it a filter function that returns `group_name==<the collision group name I want to remove>`. And I define collision group names in the URDF  by adding a group to the collision tag: `<collision group=<the collision group name I want to remove>>`. But I get a segmentation fault whenever my filter function does not always return false. The segmentation fault happens in `drake::multibody::collision::Element::CanCollideWith(drake::multibody::collision::Element const*) const ()`

After talking to Andres Valenzuela about it, the problem appears to be a bug in the bullet wrapper.

If you want to look at my code, here is my fork (It is up to date with Drake master at the time of making this issue): https://github.com/TristanThrush/drake/tree/development

The relevant code is in /drake/examples/bhpn_drake_interface/src/simulators/pr2/pr2_simulator.cc from lines 55 to 59. If you change the filter function to return false, then you should no longer get segmentation faults.

You should hopefully be able to build the relevant code by entering:
bazel build //drake/examples/bhpn_drake_interface:pr2_simulator

And you should be able to run the code by navigating to: drake/examples/bhpn_drake_interface/src/simulators/pr2/
and entering:
./example.sh

If there are any problems building my code, then let me know. It is not very clean right now and nobody else has built it recently.

SeanCurtis-TRI, 2017-08-09:
I'll take a look.

TristanThrush, 2017-08-09:
Thanks for anybody that looks at this. Just so you know, Andres said that he would look at this issue too - I am just unable to assign him.

SeanCurtis-TRI, 2017-08-09:
@avalenzu  and I can duke it out.  :)

SeanCurtis-TRI, 2017-08-09:
How urgent would you say this is?

TristanThrush, 2017-08-09:
It's preventing me from providing features in a pull request that I discussed with Professor Tedrake, Siyuan, and Naveen. So for that pull request it is fairly urgent. But I can always work on other things in the meantime.

SeanCurtis-TRI, 2017-08-09:
Assignment moved over the @avalenzu .

PR #6812 (2017-08-10): Fix removal of elements in BulletModel.
Resolves #6803. Previously, the corresponding `btCollisionObject`s were
not removed from the Bullet collision world. This led to segfaults in
the collision callback, as those `btCollisionObject`s held pointers to the
(now deleted) `drake::collision::Element` objects.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6812)
commit 4da0dbf8 (2017-08-10): Fix removal of elements in BulletModel.

Resolves #6803. Previously, the corresponding btCollisionObjects were
not removed from the Bullet collision world. This led to segfaults in
the collision callback, as those btCollisionObjects held pointers to the
(now deleted) drake::collision::Element objects.
closed 2017-08-10 by avalenzu