RobotLocomotion/drake #6804: Employ RefPtr<> mechanism to enable mutable Ref<> instances to take `&foo` as writeable arguments.

EricCousineau-TRI, 2017-08-09:
Per @jwnimmer-tri and @soonho-tri's conversation [in this PR](https://github.com/RobotLocomotion/drake/pull/6755#issuecomment-321128147), it would be useful to wrap `Eigen::Ref<>` in some way such that using it as a writeable argument does not violate GSG.

Hopefully this might address the [`const_cast<>` hack](https://eigen.tuxfamily.org/dox/TopicFunctionTakingEigenTypes.html) that Eigen's documentation suggests for mutable block references. If not, we should possibly think of alternatives.

This could possibly go in `drake/common/eigen_types.h`.

jwnimmer-tri, 2017-08-10:
You can probably fix it by guiding the type conversion with a function call.  So the call site would look something like:
```c++
DecomposeAffineExpressions(M_ * x_ + v_, x_, address_of(M_expected.block(0, 0, 3, 3)),
```
where `address_of` turns a mutable `BlockXpr` mutable reference into a `RefPtr` value.

soonho-tri, 2017-08-10:
It doesn't work as expected because `std::addressof(arg)` expects `arg` to be an l-value.

```
drake/common/test/symbolic_decompose_test.cc:131:30: 
error: no matching function for call to 'addressof'
                             std::addressof(M_expected.block(0, 0, 3, 3)),
                             ^~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__functional_base:85:1: 
note: candidate function [with _Tp = Eigen::Block<Eigen::Matrix<double, 4, 4, 0, 4, 4>, -1, -1, false>] 
not viable: expects an l-value for 1st argument
addressof(_Tp& __x) _NOEXCEPT
^
1 error generated.
```

Is it the case that we need C++17? http://en.cppreference.com/w/cpp/memory/addressof

EricCousineau-TRI, 2017-08-10:
Perhaps Jeremy meant making a custom `address_of` that just shovels whatever `MatrixBase<>` into a `RefPtr<>`?

I will see what general philosophy is behind `-Waddress-of-temporary` (and how it differs from `-Wreturn-local-addr`), to see if this is actually an appropriate error (given that's r-value, I'd think that it's lifetime would be at least that of the called function).

Perhaps the workaround would be:
```c++
auto block_xpr = X.block(...);
mutate(&xpr);
```
(Using `auto` because the resultant type is complex and noisy.)

jwnimmer-tri, 2017-08-10:
Yes, I meant our own function, not `std::addressof`.  And I think the two-line version is a fine solution also, if the developers who want this feature like it.

soonho-tri, 2017-08-10:
I see. Thanks for the clarifications. For now, I'll settle with the two-line version. I'll add documentations and open a PR.

@EricCousineau-TRI , could you improve it later by introducing `address_of`? 

I have the last question here. What'd be a good name for this `Ptr` class?
 - `drake::EigenPtr`? or
 - `drake::Ptr`
 - `Eigen::Ptr`
 - `drake::Eigen::Ptr`

EricCousineau-TRI, 2017-08-10:
@soonho-tri Yup, using `address_of` could improve.
I'd prefer something shorter, like `wref`, to indicate it's for writing, not necessarily for maintaining a reference outside the scope of the call - however, we do not disambiguate against this with out existing standards, so maybe `address_of` would work.

I think `Ptr` is too generic (and I guess `RefPtr<>` outside of Eigen is as well), and we should avoid injecting anything into Eigen's namespace.
Possibly `EigenPtr`?

jwnimmer-tri, 2017-08-10:
Yes, I was going to vote for `drake::EigenPtr` as well.  It seems the most plausible parallel to `Eigen::Ptr` without invading their namespace.

soonho-tri, 2017-08-10:
OK. I'll go with `drake::EigenPtr`. Thank you all!

sherm1, 2017-08-10:
Sorry, I just saw this. Do I understand correctly that the call site must use `address_of` rather than `&`. If so that would appear not to achieve the stated goal in the PR title, since `address_of` does not begin with `&`. That wouldn't provide the pattern-matching benefit that is (I believe) the justification for GSG's rule.

jwnimmer-tri, 2017-08-10:
`address_of` is a hell of lot better than nothing at all.  If someone can find a syntax that makes `&` then more power to them, but until then a function name will have to do.

sherm1, 2017-08-10:
It's not obvious to me why "address_of" is better than nothing at all. Neither provides the desired `&` pattern. This seems like inventing yet another scheme for marking output parameters, something that I think was a marginal benefit in the first place as evidenced by the many cases where we pass a writable pointer straight through from a function argument to an internally called function, losing the `&`. If we care that much we should insist on something like this:
```c++
// Function f has an output parameter that gets filled in by function g:
void f(int* something_ptr) {
  int& something = *int_ptr;
  g(&something);  // not g(something_ptr)
}
```
That would squeeze more utility out of the GSG pattern with little work but we don't do that, so I'm skeptical that we need to invent another mechanism to convey the concept elsewhere.

jwnimmer-tri, 2017-08-10:
Note that, as he states above, Soonho's first PR is not going to use or need `address_of` at all.  `drake::EigenPtr` precisely meets the need of enforcing `&` at call sites as if it were a regular object.

jwnimmer-tri, 2017-08-10:
@soonho-tri Can you post a WIP branch here, so we can argue about code instead of ephemera?

soonho-tri, 2017-08-10:
Here is the branch: https://github.com/soonho-tri/drake/tree/pr-add-ptr

jwnimmer-tri, 2017-08-10:
Thanks, that matches my expectations.

If Sherm feels strongly and wants to nix `address_of`, I would go along with that.  My important desire is for function signatures to declare `EigenPtr` outputs.  If the answer is to save the `VectorBlock` into a named `auto` local or to change the warning flag, instead of `address_of`, I will leave up to the folks who have to write more of the call sites than me (Eric, Soonho, Sherm, etc.).

sherm1, 2017-08-10:
Thanks @soonho-tri. `EigenPtr` looks very nice. Sorry I can't say the same for `address_of`! I'd skip that and use a local auto to permit the `&` instead, or just forgo it in that case (maybe add a comment).

PR #6852 (2017-08-22): Add EigenPtr
Close #6804

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6852)
closed 2017-08-22 by soonho-tri