RobotLocomotion/drake #6809: RigidBodyPlant::DoCalcTimeDerivative does not solve the linear system efficiently/correctly

hongkai-dai, 2017-08-10:
In RigidBodyPlant::DoCalcTimeDerivative, it computes the solution to the following problem
```
[H -Jᵀ] * [v̇] = [g]
[J  0 ]   [λ]   [h]
```
It solves this problem by calling `LinearSystemSolver`, which does a SVD decomposition of the matrix on the left hand-side.

This approach has two problems

1. SVD is not the most efficient approach to this problem. It has complexity O(m + n)³, where `m` is the number of rows in `H`, and `n` is the number of rows in `J.` Instead, we should solve the following problem
   ```
   JH⁻¹Jᵀ * λ = h - JH⁻¹g
   ```
   which is smaller in size. And since `H` is positive definite, we can solve this linear system a lot more efficient by using Cholesky decomposition. We did this in solving the KKT condition of QP.
2. SVD does not give the meaningful solution, when the linear problem is inconsistent. In this case, an SVD solver to `A * x = b` tries to find the solution that minimizes the error `|b - A* x|²`. But here the term `b = [g; h]`, which does not have the same unit.

cc @edrumwri @sherm1 .

I am working on adding a constraint type `KKTconditionLinearEqualityConstraint`, that encodes the linear equality constraint
```
[H -Jᵀ] * [x] = [g]
[J  0 ]   [λ]   [h]
```
and a special solver for this problem (mostly taken from EqualityConstraintQPSolver.)

sherm1, 2017-08-10:
:+1:
Some minor thoughts:
- Slightly nicer IMO to redefine the sign of λ to preserve symmetry of the matrix:
    ```
    [H Jᵀ] * [x] = [g]
    [J 0 ]   [λ]   [h]
    ```
- When we have force constraints like sliding friction or softened contact constraints, the "0" in the matrix gains some terms:
    ```
    [H Jᵀ] * [x] = [g]
    [J D ]   [λ]   [h]
    ```
    Then the non-SPD part of the solution becomes
    ```
    (JH⁻¹Jᵀ-D) * λ = JH⁻¹g - h
    ```
     That isn't symmetric any more but it doesn't matter for the linear algebra because QR, QTZ, and SVD don't care.

    I'm just mentioning that in case it is just as easy for you to allow for a non-zero lower right block.

hongkai-dai, 2017-08-10:
Since we are going to migrate to MultiBodyPlant, it doesn't worth fixing this.

closed 2017-08-10 by hongkai-dai

RussTedrake, 2017-08-13:
@hongkai-dai -- i would have thought that the fix would also be used in multibody plant.  and we have enough users of rigidbodyplant now that it's still useful?