RobotLocomotion/drake #6827: Add a LCM log playing system block.

siyuanfeng-tri, 2017-08-11:
The initial plan is to implement a system block for playing back LCM logs that is functionally very much like the LcmSubscriber. The block will be emitting a single channel with a known type. This is necessary for running tests on CI based on logs from hardware experiments.

siyuanfeng-tri, 2017-10-03:
A sanity design check before I polish it. Also cc @RussTedrake @EricCousineau-TRI can you take a quick look at the system level example (drake/systems/lcm/test/lcm_log_playback_system_test.cc), see if you are happy with it. 
https://github.com/siyuanfeng-tri/drake/commit/19817cb80e1d15aa7c3e2db5e751fac87f6a7682

The overall goal is to minimize code change on the user end (e.g. different diagram building, etc) to substitute real lcm traffic with a log ( for both subscribe / publish). The basic idea is to provide another DrakeLcmLog : public DrakeLcmInterface. For publishers, the diagram stays exactly the same, for subscribers, I need to add an additional LcmLogPlaybackSystem to the diagram unfortunately. The reason being I need to get the simulator's time from the context to dispatch the message at the exact moment.

RussTedrake, 2017-10-03:
nice!  the basic design seems reasonable to me.  
if you really want to enable the desired workflow (caller changes the DrakeLCMInterface, but not the diagram), then perhaps you could introduce an additional method to DrakeLCMInterface that gets called from LCMSubscriber::DoCalcNextUpdateTime and does nothing for DrakeLcm and DrakeMockLcm, but does what you need for the DrakeLcmLog?

siyuanfeng-tri, 2017-10-03:
great!
i think i can get rid of the extra block, and hide the event advancing part in lcm subscriber. it might become a bit more arcane though. doing it in a separate block is conceptually more clear. i was going to add some utility method to wrap a diagram with the extra block. let me try both first.

jwnimmer-tri, 2019-03-04:
@siyuanfeng-tri What's your plan for this?  Did it merge already?  Should we close this?

jwnimmer-tri, 2019-10-30:
We have https://github.com/RobotLocomotion/drake/blob/master/systems/lcm/lcm_log_playback_system.h on master.  I think that is what this issue was about?  Please re-open an clarify, if not.

closed 2019-10-30 by jwnimmer-tri