RobotLocomotion/drake #6829: error: invalid operands to binary expression ('drake::symbolic::Expression' and 'drake::symbolic::Expression')

tarquasso, 2017-08-11:
```
./drake/examples/DoublePendulum/gen/double_pendulum_state_vector.h:81:16: error: invalid operands to binary expression ('drake::symbolic::Expression' and 'drake::symbolic::Expression')
  decltype(T() < T()) IsValid() const 
```
Having troubles compiling a new plant using the auto-generated state vector file:  
[`gen/double_pendulum_state_vector.h` ](https://github.com/tarquasso/drake/blob/double_pendulum/drake/examples/DoublePendulum/gen/double_pendulum_state_vector.h) file (It was generated with: [`double_pendulum_gen.sh`](https://github.com/tarquasso/drake/blob/double_pendulum/drake/examples/DoublePendulum/double_pendulum_gen.sh))

A comparision `T() < T()` fails for two `drake::symbolic::Expression`.

The example is called "DoublePendulum", a duplicate of Acrobot example, but fully actuated. See the other source files here:
[Github DoublePendulum Example](https://github.com/tarquasso/drake/tree/double_pendulum/drake/examples/DoublePendulum)
(branched of master sha `b6a35cece18a34f06ece2fdc24b15096a474ee9a` as of 8/11/17 10am)

Here the full compiler error message:
```
ERROR: /home/rkk/drake-distro/drake/examples/DoublePendulum/BUILD:64:1: C++ compilation of rule '//drake/examples/DoublePendulum:double_pendulum_plant' failed: Process exited with status 1 [sandboxed].
In file included from drake/examples/DoublePendulum/double_pendulum_plant.cc:1:
In file included from ./drake/examples/DoublePendulum/double_pendulum_plant.h:5:
./drake/examples/DoublePendulum/gen/double_pendulum_state_vector.h:81:16: error: invalid operands to binary expression ('drake::symbolic::Expression' and 'drake::symbolic::Expression')
  decltype(T() < T()) IsValid() const {
           ~~~ ^ ~~~
drake/examples/DoublePendulum/double_pendulum_plant.cc:40:33: note: in instantiation of template class 'drake::examples::double_pendulum::DoublePendulumStateVector<drake::symbolic::Expression>' requested here
  this->DeclareVectorOutputPort(DoublePendulumStateVector<T>());
                                ^
drake/examples/DoublePendulum/double_pendulum_plant.cc:160:14: note: in instantiation of member function 'drake::examples::double_pendulum::DoublePendulumPlant<drake::symbolic::Expression>::DoublePendulumPlant' requested here
  return new DoublePendulumPlant<symbolic::Expression>();
             ^
./drake/common/symbolic_expression.h:63:6: note: candidate function not viable: no known conversion from 'drake::symbolic::Expression' to 'drake::symbolic::ExpressionKind' for 1st argument
bool operator<(ExpressionKind k1, ExpressionKind k2);
     ^
1 error generated.
```
Any tips on how to understand this / fix it are highly appreciated!

soonho-tri, 2017-08-11:
Could you test the following patch?

```diff
--- a/drake/examples/DoublePendulum/double_pendulum_plant.h
+++ b/drake/examples/DoublePendulum/double_pendulum_plant.h
@@ -7,6 +7,7 @@
 #include "drake/systems/framework/diagram.h"
 #include "drake/systems/framework/leaf_system.h"
 #include "drake/systems/primitives/affine_system.h"
+#include "drake/common/symbolic_formula.h"
```

soonho-tri, 2017-08-11:
I think a better solution is to rebase your branch against the latest `master` branch of `RobotLocomotion/drake` and to include the following in `double_pendulum_plant.h` if needed:

```c++
#include "drake/common/symbolic.h"
```

tarquasso, 2017-08-14:
I performed the rebase to current master, now I can't build anymore:

```
~/drake-distro/drake/examples/DoublePendulum$ bazel build ...
ERROR: /home/rkk/.cache/bazel/_bazel_rkk/cc1738dde3d5a7f5de70cb525c89ee8a/external/drake/tools/install.bzl:278:9: name 'DefaultInfo' is not defined.
ERROR: error loading package 'tools': Extension 'tools/install.bzl' has errors.
INFO: Elapsed time: 0.305s
```

and my externals folder is filled with nested roots:
```
git status
On branch double_pendulum
Your branch is up-to-date with 'origin/double_pendulum'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   externals/bullet (new commits)

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	drake/examples/LittleDog/
	externals/avl/
	externals/ccd/
	externals/fcl/
	externals/gurobi/
	externals/iris/
	externals/meshconverters/
	externals/mosek/
	externals/octomap/
	externals/sedumi/
	externals/signalscope/
	externals/snopt/
	externals/spotless/
	externals/vtk/
	externals/xfoil/
	externals/yalmip/
```
Shall I clean my workspace according to: [Clean Workspace Instruction for make](http://drake.mit.edu/from_source.html#how-to-clean-your-workspace)
or is there a better way since I am using bazel?

jwnimmer-tri, 2017-08-14:
There is a `bazel clean` command, but I don't think that's your problem.  In http://drake.mit.edu/developers.html, the Bazel version we use is 0.5.2 (and I believe 0.5.3 also works); is that what you're trying to use?

tarquasso, 2017-08-15:
updating bazel and rebasing from master fixed the issue.

closed 2017-08-15 by tarquasso