RobotLocomotion/drake #6836: AddLinearConstraint(Formula) cannot accept limits at infinity

RussTedrake, 2017-08-14:
If a user passes in `inf` or `-inf` into as part of a symbolic formula, the resulting constraints will have `lb` or `ub = NaN`.  This happens in 
https://github.com/RobotLocomotion/drake/blob/master/drake/automotive/test/trajectory_optimization_test.cc#L52
and happens because of the logic here:
https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/create_constraint.cc#L127
(which ultimately results in `-inf + inf = nan` or `inf - inf = nan`).

fyi @jadecastro -- this is very possibly the reason why ipopt was performing so badly on the trajectory optimization example with simple cars.  it's probably worth trying ipopt again once this is resolved.

RussTedrake, 2017-08-14:
my mistake.  it doesn't happen on master (i had local mods).  but it can be reproduced in the automotive example by inserting at the location above
```
prog.AddLinearConstraint(lower_limit.get_value() <= prog.input(0));
prog.AddLinearConstraint(prog.input(0) <= upper_limit.get_value());
```

soonho-tri, 2017-08-14:
@RussTedrake , I'm working on the fix. Thanks for the report.

soonho-tri, 2017-08-16:
> fyi @jadecastro -- this is very possibly the reason why ipopt was performing so badly on the trajectory optimization example with simple cars. it's probably worth trying ipopt again once this is resolved.

@jadecastro , this is issue is fixed in `master`. Could you test it with `ipopt`? Thank you.

closed 2017-08-16 by soonho-tri

jwnimmer-tri, 2017-08-16:
@soonho-tri When running unit tests (such as automotive trajectory optimization) in debug mode, if we are computing with `NaN`s then we should have been failing-fast, not taking a stupid-long time.  If indeed that's what was happening, please open an issue to add `DRAKE_ASSERT(!std::isnan(stuff))` to enough places that we would have caught this earlier.

soonho-tri, 2017-08-16:
I see. 

I've added `DRAKE_DEMAND(!std::isnan(...))` in #6849. I will check the other files and add more.

RussTedrake, 2017-08-16:
My apologies.  The nan issue was not in master before (see my clarification above).  But indeed, it was not failing fast on my branch... it took some digging to discover it.

soonho-tri, 2017-08-16:
Not a problem. I believe that the bomb was there and it was just a matter of time to trigger it. I believe that the symbolic code under `common` directory is good, but maybe the code under `solvers` is not as good. As @jwnimmer-tri suggested, I'll inspect them asap.

jadecastro, 2017-08-16:
I compared pre- and post-PR commits the test using IPOPT and there's absolutely no change to resulting trajectory, which I think is expected.
(I did not try to recreate the result when replacing `AddInputBounds()` with `AddLinearConstraint()`, however.)

soonho-tri, 2017-08-16:
Good to know. Thanks for the check, @jadecastro !

PR #6849 (2017-08-16): Fix ParseLinearConstraint to handle infinity
Fix #6836.

Given a symbolic formula `e1 <= e2`, we want to construct a linear constraint by finding its lower and upper-bounds. So far, we have used `-∞ <= e1 - e2 <= 0` but it does not work as expected if we have `∞` in `e1` or `e2`.

For example, consider a simple constraint `x <= ∞`. Using the existing approach, we have:
```
    -∞ <= x - ∞ <= 0
    lb      v     ub
```
and we subtract the constant term `c = -∞` of `v` from `lb` and `ub` to have:

```
    -∞ + ∞  <=    x    <= -∞
    lb - c      v - c    ub - c
```
That's where we have `nan` ( = [... cut]