RobotLocomotion/drake #6839: Fix Style Guide violations for `evaulator_base`, `constraint`, and `cost` targets (and downstream dependencies)

EricCousineau-TRI, 2017-08-14:
As follow-up to a conversation with @sherm1 in #6818, this issue will address fixing GSG violations present:

- [ ] Usage of rvalue-references for output arguments. (contingent upon a agreed-upon convention for #6804)
- [ ] Name conventions for `detail::FunctionTraits<>` (e.g., `numInputs()`, `numOutputs()`)
- [ ] Inconsistent interfaces between `size_t` and `int` (e.g. `num_constraints()` and its usages in the solver interface implementations).

This will be implemented when there are no significant `MathematicalProgram` PRs in flight.

\cc @hongkai-dai @soonho-tri

soonho-tri, 2017-08-14:
Great! I'll open the `drake::EigenPtr` PR today. Thanks for the patience.

jwnimmer-tri, 2021-11-11:
#8349 is somewhat related.