RobotLocomotion/drake #6842: physics of Acrobot changed, finding short trajectory for swing up is broken

tarquasso, 2017-08-14:
Discussed in issue #6840, recovering the simulation of the acrobot swing up (`acrobot_run_swing_up_traj_optimization.cc`) was fixed by commenting out changes to the IntegratorBase.h file. 

This only solves part of the problem with the acrobot swing up. The other issue is that while previously it was possible to find a swing up trajectory that takes 4.69 seconds, it is now not able to find a trajectory that finished before the maximum parameter `kTrajectoryTimeUpperBound` and that has to be at least 10 seconds.

[Videos of swing ups with old and new physics](https://www.dropbox.com/sh/n567felaqmhwkty/AADg0iX1h26l_6-NwvaOPvhLa?dl=0)

Running git bisect leads to:
[Commit that changes AcrobotPlant Parameters](https://github.com/RobotLocomotion/drake/commit/cf58bc3c708c78546227f193ed75df2d173658a4)
```
cf58bc3c708c78546227f193ed75df2d173658a4 is the first bad commit
commit cf58bc3c708c78546227f193ed75df2d173658a4
Author: Jeremy Nimmer <jeremy.nimmer@tri.global>
Date:   Wed Jul 19 10:32:09 2017 -0400
AcrobotPlant retains parameters during transmogrification
```

This was found by setting upper bound parameter in `acrobot_run_swing_up_traj_optimization.cc` to:
kTrajectoryTimeUpperBound = 5;

and adding print out to l.67 of `acrobot_run_swing_up_traj_optimization.cc`:
```
std::cout <<"End Time: " << pp_xtraj.get_end_time()<< std::endl;
```

Also, fixed the problem from #6840 by commenting out:
```
diff --git a/drake/systems/analysis/integrator_base.h b/drake/systems/analysis/integrator_base.h
index a3b6374..62a9b71 100644
--- a/drake/systems/analysis/integrator_base.h
+++ b/drake/systems/analysis/integrator_base.h
@@ -1645,11 +1645,11 @@ typename IntegratorBase<T>::StepResult IntegratorBase<T>::IntegrateAtMost(
 
   // If there is no continuous state, there will be no need to limit the
   // integration step size.
-  if (get_context().get_continuous_state()->size() == 0) {
-    Context<T>* context = get_mutable_context();
-    context->set_time(context->get_time() + dt);
-    return candidate_result;
-  }
+//  if (get_context().get_continuous_state()->size() == 0) {
+//    Context<T>* context = get_mutable_context();
+//    context->set_time(context->get_time() + dt);
+//    return candidate_result;
+//  }
```
Please assign @hongkai-dai or @jwnimmer-tri to this

jwnimmer-tri, 2017-08-14:
I suspect that my "AcrobotPlant retains parameters during transmogrification" commit merely exposed some preexisting underlying problem, but @hongkai-dai please ping me again if otherwise.

amcastro-tri, 2017-08-15:
I was able to reproduce the problem reported by @tarquasso with `IntegratorBase`. There is a bug here introduced with #6593. The integrator is in "fixed step size mode" however it erroneously takes a single step size to the end of the simulation because it detected the system has no state. In the acrobot example this is the case since the "plant" is simply a `TrajectorySource` with no state. It is just playing back the computed trajectory.
Therefore, there are two different bugs with this example:
1. The one bug identified by @hongkai-dai in #6845 where we switched the default plant's constructor to using the `CreateAcrobotMIT()`. This changed the parameters of the plant and therefore the different end times observed by @tarquasso in his simulations.
2. The changes introduced in [integrator base](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/analysis/integrator_base.h#L1648) by PR #6593. Notice also @edrumwri that that change causes the integrator to leave without updating the statistics as done below for the other cases. Something to keep in mind when fixing integrator base.

amcastro-tri, 2017-08-15:
Created #6846

amcastro-tri, 2017-08-16:
Fixed by #6845

closed 2017-08-16 by amcastro-tri