RobotLocomotion/drake #6846: IntegratorBase steps to the end of the simulation for systems with no continuous state.

amcastro-tri, 2017-08-15:
This problem was found by @tarquasso in #6842. I'll summarize here the problem relevant to `IntegratorBase`.

There is a bug here introduced with #6593 at [this particular location in integrator base](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/analysis/integrator_base.h#L1648). The integrator is in "fixed step size mode" however it erroneously takes a single step size to the end of the simulation because it detected the system has no state. In the acrobot example this is the case since the "plant" is simply a `TrajectorySource` with no state. It is just playing back the computed trajectory.

Notice also @edrumwri that that change causes the integrator to leave without updating the statistics as done below for the other cases. Something to keep in mind when fixing integrator base.

To see the problem with the acrobot example (acrobot_run_swing_up_traj_optimization.cc) please add prints to see, after simulation:
- simulator.get_context().get_time());
- simulator.get_num_steps_taken());
- simulator.get_integrator()->supports_error_estimation());
- simulator.get_integrator()->get_fixed_step_mode());
- simulator.get_integrator()->get_maximum_step_size());
- simulator.get_integrator()->get_largest_step_size_taken());
- simulator.get_integrator()->get_num_steps_taken());
- simulator.get_integrator()->get_num_step_shrinkages_from_error_control());
- simulator.get_integrator()->get_num_step_shrinkages_from_substep_failures());

You will observe the following problems:
- First of all, the visualization does not advance in time in the drake visualizer.
- simulator.get_num_steps_taken() = 2
- simulator.get_integrator()->get_largest_step_size_taken() = 9.9999
- simulator.get_integrator()->get_num_steps_taken() = 2

When commenting out the entire if statement [here](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/analysis/integrator_base.h#L1648) we get the expected behavior:
- simulator.get_num_steps_taken() = 10001
- simulator.get_integrator()->get_largest_step_size_taken() = 0.001
- simulator.get_integrator()->get_num_steps_taken() = 10001

amcastro-tri, 2017-08-15:
Interesting super relevant discussion started by @sherm1 [here](https://github.com/RobotLocomotion/drake/pull/6593#issuecomment-316224617). @sherm1 saw this change as a potential problem however per f2f (not transcribed into Reviewable) the change made it into master.

amcastro-tri, 2017-08-15:
Per f2f with @sherm1 and @edrumwri, we concluded that the `Simulator` is actually doing the right thing; it sees a system with no continuous state and therefore decides on not calling the integrator (which is the one guy in charge of continuous integration) at all.
We concluded also that there should be someone asking for a publishing rate in the particular case of the acrobot (#6840) and that it makes that someone to be the `BotVisualizer`. Therefore we will add a second argument to `BotVisualizer`'s constructor specifying the publishing rate (with a default value of 60 Hz for most commonly occurring Drake examples).
I will push a fix PR soon with this proposal.

tarquasso, 2017-08-15:
For a user, it seems not intuitive having to come up with having to define a publishing rate as @amcastro-tri proposes it as the fix for now:
```
publisher->set_publish_period(1.0 / 60.0);
```

I would propose to that a parameter should be asked for somewhere, possibly when setting up `state_source` by having a second argument, see:
```
const PiecewisePolynomialTrajectory pp_xtraj = dircol_traj.ReconstructStateTrajectory();
auto state_source = builder.AddSystem<systems::TrajectorySource>(pp_xtraj,publishing_rate);
```
that argument can have a default value. 

Or alternatively, add it as an argument to
```
auto publisher = builder.AddSystem<systems::DrakeVisualizer>(*tree, &lcm,publishing_rate);
```
Probably adding it as an argument to the `TrajectorySource` makes more sense. This way, if the integrator is not stepping forward due to not having a continuous state, the publisher can still check if the source has defined a publishing rate. What do you think?

sherm1, 2017-08-15:
> Probably adding it as an argument to the TrajectorySource makes more sense. 

I don't think so. The source normally doesn't even have a Publish method. OTOH the drake visualizer has to publish LCM messages. Better to have it pull computations as needed.

SeanCurtis-TRI, 2017-08-15:
I think this is partially muddied by the fact that we have things (currently) configured to publish on every integrator step. That's also problematic. It takes thinking of publishing out of the hands of the user. Publishing frequency *should* be in the purview of the programmer assembling a system.

amcastro-tri, 2017-08-15:
so do you think `DrakeVisualizer`'s constructor should ask for a publishing rate **always**?

SeanCurtis-TRI, 2017-08-15:
I think `DrakeVisualizer` could have a default value (and making it 60 Hz is rational). I just don't think the `Simulator` should have a "publish on every integration step" default. Ultimately, we want systems with publish (or other discrete behaviors) to have that property out front and common. People shouldn't forget that their system works because it publishes. Does that make sense?

sherm1, 2017-08-15:
> I just don't think the Simulator should have a "publish on every integration step" default.

I believe that feature is an anachronism since Siyuan made it so that Systems could schedule their Publish calls that way. We should remove it at some point but there are some unknown number of tests and examples that are using it making it a bit sticky.

amcastro-tri, 2017-08-16:
I'd like to close this issue unless someone has a better proposal for this

edrumwri, 2017-08-17:
Seconded.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

closed 2017-08-18 by amcastro-tri