RobotLocomotion/drake #6853: divide_allowing_by_zero warning on GCC 4.8

jwnimmer-tri, 2017-08-16:
When building with GCC 4.8, we see this ...
```
In file included from
external/drake/drake/systems/analysis/implicit_euler_integrator.cc:2:0:
external/drake/drake/systems/analysis/implicit_euler_integrator-inl.h:411:49:
warning: 'no_sanitize_undefined' attribute directive ignored [-Wattributes]
T divide_allowing_by_zero(const T& n, const T& d) {
```
... because that attribute is new as of GCC 4.9.

While Drake doesn't support GCC 4.8 officially in CI, we are keeping it clean on Trusty's GCC 4.8 for a short while longer, internally at TRI.  Please update the preprocessor guards on this line so that the attribute is only enabled for GCC >= 4.9.

/CC @stonier

jamiesnape, 2017-08-16:
> While Drake doesn't support GCC 4.8 officially in CI, we are keeping it clean on Trusty's GCC 4.8 for a short while longer, internally at TRI.

This has never been formally documented. Please do so.

jwnimmer-tri, 2017-08-16:
The hotfix is done, but there is no documentation yet per the prior comment.  Re-assigning to @stonier for assessment.

jwnimmer-tri, 2017-09-21:
Sigh.  #7070 was sad, though perhaps the last time we'll need it... for the record, we have less than two weeks left of GCC 4.8 support from the TRI side.

PR #7158 (2017-10-03): Remove support for Ubuntu 14.04 (Trusty)
Toward #5012. Closes #4913. Closes #6853.

For CI testing at the moment. I will open a release branch for Trusty before this is merged.

(Failures are expected for Trusty builds, for obvious reasons.)

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7158)
closed 2017-10-03 by jamiesnape