RobotLocomotion/drake #6855: `bazel run` blocks other `run` and `build` commands

rdeits, 2017-08-16:
This came up while I was trying to run the visualizer, simulator, and valkyrie controller. Doing `bazel run <simulator>` locks the entire Bazel system, meaning that no other command (like `bazel run <controller>`) can be used at the same time. 

The workaround is to dig around inside the `bazel-bin` folder and run the actual binaries. Is that the intended workflow?

Some discussion of this limitation: https://groups.google.com/forum/#!topic/bazel-discuss/mS98nXz9PbY

EricCousineau-TRI, 2017-08-16:
That is presently the intended workflow, AFAIK.

And thank you for posting that discussion! It looks like we might want to adopt something akin to `bazel-run.sh`:
https://github.com/bazelbuild/bazel/blob/master/scripts/bazel-run.sh
Only caveat is, it's not clear how changing the PWD the `runfiles` directory may affect passing in arguments, e.g.:
```
./tools/bazel-run.sh //tools:drake-visualizer --script ./drake/multibody/rigid_body_plant/visualization/show_frames.py
```
Having to incorporate `show_frames.py` into the dependencies seems like it'd be quite a pain...
Will briefly try this out to see how this works.

jwnimmer-tri, 2017-08-17:
My inclination would be to stick with running `bazel-bin/foo/bar` directly.

EricCousineau-TRI, 2018-12-08:
This is what we've been doing, seems fine. Closing.

closed 2018-12-08 by EricCousineau-TRI