RobotLocomotion/drake #6878: Need a method to check if a system has ONLY a single periodic discrete time update

RussTedrake, 2017-08-20:
Many of the analysis tools we have / will write for discrete time systems assume the system is governed by (only) simple difference equations (e.g. `x[n+1] = f(n,x[n],u[n])`).  It's easy to register a system that has this property, but I don't know how to robustly check if a `System` class has this simple form.  

I *think* it should be a relatively straight-forward method to write in `System` which just checks that we have a single periodic event, and no other types of events?  But I'm hoping someone more familiar with the event handling can write this (both to get the implementation right now, but also to protect it against future changes).

N.B. I'll also need to be able to get the duration of that periodic update.  

Tagging @edrumwri , but feel free to reassign as appropriate.

edrumwri, 2017-08-21:
Russ, we can identify whether a system has discrete variables, a periodic
update trigger, and a discrete update event, but this may undesirably
identify some systems as not modeled solely by difference equations.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

RussTedrake, 2017-08-21:
I think what is required is identifying when a system does NOT have other types of discrete events?  is it possible to check all of the events (i think i saw at least two lists implemented right now?) and ask if they are only these simple periodic events?

edrumwri, 2017-08-21:
I'm not sure exactly what you're asking, so I'll attempt to disambiguate:

The language we have in place is "triggers" are the reason that an event
should occur and "event handlers" are the pieces of code that are called
when a trigger is, ahem, triggered.

We can determine whether a periodic timer triggers a discrete update event
handler, which should cover some number of the simple systems you
described. However, it is also possible to have a periodic trigger
instigate an unrestricted update handler (which could organically model the
same system); another way to effect this kind of system would be to have a
witness function over time instigate a discrete update handler, which then
handles the update. There are many ways to skin this cat. How many of them
do you want to capture?

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

RussTedrake, 2017-08-23:
Thanks for the clarification.  What I'm asking for is to reflect at the system level about the number and variety of triggers that have been registered for the system.  The special case I care about is identifying systems that will trigger **only** the discrete update event and **only** on a single periodic update.  Again, I also need to extract the period of that update.  

The first specific use case is captured in TODO's in #6887.

Happy to f2f today if that's easier.

RussTedrake, 2017-08-23:
specific relevant TODOs:
https://github.com/RobotLocomotion/drake/pull/6887/files#diff-a8bece3c210e4efd988288e19dd0d793R99
https://github.com/RobotLocomotion/drake/pull/6887/files#diff-a8bece3c210e4efd988288e19dd0d793R108
https://github.com/RobotLocomotion/drake/pull/6887/files#diff-fd06567d5d2f83c9e1b8a6b42805ca8fR50
(apologies if the links to the diffs prove to be fragile)

edrumwri, 2017-08-23:
Ok, that tells me what I need to know. I'll get to work on this. Please let
me know if you need it done on a specific deadline.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

edrumwri, 2017-12-22:
Addressed by #7105.

closed 2017-12-22 by edrumwri