RobotLocomotion/drake #6890: Create new doxygen function for eigen scalar template boilerplate

SeanCurtis-TRI, 2017-08-23:
When a class is templated on eigen scalars, we get boilerplate that looks something like this:

/// @tparam T The scalar type. Must be a valid Eigen scalar.
///
/// Instantiated templates for the following kinds of T's are provided:
/// - double
/// - AutoDiffXd
///
/// They are already available to link against in the containing library.
/// No other values for T are currently supported.

It would be good if this could be supplanted with one or more custom doxygen commands that would make this documentation uniform.

Ideally, it would be great if you could do `@eigen_scalar{double, AutoDiffXd}`, however, [doxygen documentation](https://www.stack.nl/~dimitri/doxygen/manual/custcmd.html) doesn't seem to allow this (the number of arguments must be fixed). 

Alternatively, we introduce a *set* of custom commands:
  - `@eigen_scalar_d`
  - `@eigen_scalar_ad`
  - `@eigen_scalar_ads`
  - `@eigen_scalar_ds`
  - (Would we have any variant *without* double? Unlikely.)

sammy-tri, 2017-08-23:
I think I like the idea...  presuming we don't think the total quantity of supported scalartypes goes beyond about five...  (/me has a horrible thought about the code which machine generates the part of the Doxyfile with hundreds of permutations of `@eigen_scalar_asdfghjq`)

amcastro-tri, 2017-08-24:
In the **Aliases with arguments** section says: _"Note that you can overload an alias by a version with multiple arguments."_ Maybe we could do that to support multiple arguments? which would imply there is a limit but I guess that if we go beyond five arguments we already lost... and of course people might opt for writing their own customized boilerplate if so they need it.

jwnimmer-tri, 2017-08-25:
I would be inclined to attempt solving this in conjunction with de-boilerplating the `.cc` file (where the instantiations are also listed out).  It's possible that a single macro in the `.cc` file would suit both needs.

jwnimmer-tri, 2020-03-05:
Relates to #12814 so I might end up doing a little work towards something here.

jwnimmer-tri, 2020-03-09:
A first pass has been merged.  Remaining work is to improve the macro with experience, and to finish adding uses of the macro to any code that we missed or skipped in the first pass.  In the `//geometry/...` folder particularly, I skipped over many items.

SeanCurtis-TRI, 2021-12-04:
This is resolved our custom `@tparam_default_scalar` and `@tparam_non_symbolic_scalar`.

closed 2021-12-04 by SeanCurtis-TRI