RobotLocomotion/drake #6892: IKTraj: setAdditionaltSamples does not take effect

avalenzu, 2017-08-23:
The constructor for `IKInbetweenConstraint` calls the `Constraint` constructor with 0 as the number of constraints. Then, on line 529 of `inverse_kinematics_trajectory_backend.cc` the `num_constraints()` method is called for the newly created `IKInbetweenConstraint` shared pointer. This will always return 0, and therefore the `IKInbetweenConstraint` object is never added to the mathematical program.

`test_ik_traj.cc` does call the `setAdditionaltSamples` method of `IKOptions`, which should exercise this functionality. However, the test only checks that the IK traj problem solved successfully. Since this bug results in constraints being omitted from the mathematical program, it actually makes the IK traj problem easier to solve.

avalenzu, 2017-08-23:
I'm working on a workaround that uses a static method to compute the number of constraints to be passed to the `Constraint` constructor.

sammy-tri, 2017-08-23:
It looks like this has probably only been broken since 5ed4f1ee...  Do we need a check in `Constraint::set_bounds()` that the side of the bounds arrays are equal to num_constraints? I think that would probably have caught this.  I think we probably lost the IK test coverage which would have caught it when we lost the MATLAB tests.

avalenzu, 2017-08-24:
I agree that 5ed4f1e is the probable cause. @EricCousineau-TRI, could you see if there's a way to restore the former functionality without undoing the results you were going for in #6818?

pchorak, 2018-07-24:
Looks like this is fixed as of PR #7892, right?

jwnimmer-tri, 2018-07-24:
@hongkai-dai Can you check?

hongkai-dai, 2019-03-19:
Since we are not going to use `IKtraj`, I will close this issue.

closed 2019-03-19 by hongkai-dai