RobotLocomotion/drake #6901: Standardize terminology for knots/time steps/samples

avalenzu, 2017-08-24:
We seem to be of many minds as to what to call the points at which the state and input values are discretized in a trajectory optimization. All of the following terms are used in different parts of the code base:
 - timestep
 - knot
 - time
 - sample

We should pick a consistent terminology use it throughout. 

I propose "knot" for the individual points. @RussTedrake and @hongkai-dai (or anyone else), what do you think we should use?

sammy-tri, 2017-08-24:
to clarify, this is referring to the value of `t` at a specific point, not the state/input values themselves?

avalenzu, 2017-08-24:
I meant this to refer to the state/input values, actually. But we should have a standard term for the times too.

SeanCurtis-TRI, 2017-08-24:
I have the vague sense from differential geometry as knots being the parameter values and not the function values. Plus, "knot" reflects the underlying mathematical representation but doesn't say much about the application domain.

RussTedrake, 2017-08-24:
i agree that there is a bit of a mess there, but those words are playing different roles:
- timestep is h.  aka dt
- time / sample time is t.  aka sum h.
- knot is x(t)

hongkai-dai, 2017-08-24:
I agree what @RussTedrake  says. Just one thing, according to Wikipedia, `knot` means `tᵢ`, instead of `x(tᵢ)`. https://en.wikipedia.org/wiki/Spline_(mathematics). So we need to add documentation in our code to explain the meaning of these terms in our code.

RussTedrake, 2017-08-24:
ah.  in that case, perhaps i have been using "knots" incorrectly for years.  sigh.  that's worth fixing. 
(both in the code and i my brain)

hongkai-dai, 2017-08-24:
I haven't found a good term for `x(tᵢ)` yet, but @SeanCurtis-TRI will look up his book for the right terminology.

SeanCurtis-TRI, 2017-08-24:
I'm suspicious there is a good name for the trajectory values.

Currently, we're doing piecewise interpolation on samples `(s₁, s₂, ..., sₙ)`. We've associated each of those values with a parameter value `(t₁, t₂, ..., tₙ)`. If we're simply doing piecewise interpolation (such that `x(tᵢ) = sᵢ`),  "knots" for `t` seems acceptable. (Although, I tend to think of knots in more of a NURBS context in which one can discuss their multiplicity and in which there's a distinction between the control points and the evaluation.) I suspect there won't be a universal name for the data points we're interpolating. But I'll check out my book when I get home to see if I can discover one (as @hongkai-dai mentioned).

avalenzu, 2017-08-24:
@RussTedrake, I agree that those words should play different roles - we just aren't using them in those roles consistently now.

RussTedrake, 2020-03-27:
@avalenzu -- hopefully #12939 resolved this.  will close.

closed 2020-03-27 by RussTedrake