RobotLocomotion/drake #6908: UBSan Failure in run_quadrotor_dynamics_test

m-chaturvedi, 2017-08-24:
`bazel test  -c  dbg --copt -O0 --config=ubsan --test_output=all //drake/examples/quadrotor:run_quadrotor_dynamics_test`

The errors seems to be due to run-time overflow when converting double to float, [here](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/rigid_body_plant/viewer_draw_translator.cc#L69).  The LCM structure has floats which are trying to store doubles.

stonier, 2017-09-21:
@naveenoid *bump* this is the last one to close out the sanitiser jobs we are setting up, do you have an ETA for us? ;)

naveenoid, 2017-09-21:
OK working on it right now. ETA today

naveenoid, 2017-09-22:
WIP!

m-chaturvedi, 2017-10-06:
Hi @naveenoid!  Did you get a chance to look at the problem?

naveenoid, 2017-10-06:
Sadly still working on it since I didnt get much coding time. This afternoon I will spend time exclusively on this. Sorry again.

m-chaturvedi, 2017-10-06:
That's alright.  Thanks!

stonier, 2017-10-12:
@naveenoid need to get this resolved soon. If we let it slide too long, then there is a good chance we will have to repeat the cycle of bringing the quickly evolving codebase again up to the point where we can consider adding the sanitisers to the buildcop purview.

jwnimmer-tri, 2017-10-12:
Did #7212 not resolve the problem?

m-chaturvedi, 2017-10-12:
It did, we forgot to close the issue.   Closed via #7212 .

closed 2017-10-12 by m-chaturvedi