RobotLocomotion/drake #6915: restrict VectorSystem to have a (single) BasicVector of Parameters

RussTedrake, 2017-08-25:
@jwnimmer-tri -- please check me on this, but 
a) i don't think this would be very restrictive, given all of the uses of parameters so far in the code.
b) i think it could do some beautiful things to the API... specifically `DoCalcVectorTimeDerivatives` and friends could change from taking `{Context, state, input}` to `{time, state, input, params}`.  For novices, the `Context` is not immediately obvious and there is a lot to bite off when you first look at it.  Also it's always felt a little yucky to have the state and inputs available both through the context and directly.

jwnimmer-tri, 2017-08-25:
Rejecting abstract parameters is a no-brainer; its a `VectorSystem` after all.  I also have no problem with limiting a `VectorSystem` to have at most a single group of numeric parameters.  (This necessarily implies that they are phrased as a single `BasicVector`.)  This seems consistent with the intent of the class, and if your experience suggests that it's a reasonable trade-off to nix multple groups of parameters, then I think we're good.

The changes to the signature are a slightly different question.  I think adding a `params` argument is fine if you're willing to tolerate an API bump.  (There are ways to avoid a flag-day for it, if you wish.)  The real question is removing access to the `Context` with no recourse.  That means, e.g., `Context::get_accuracy` is no longer available to the user's subclass code, nor would the subclass code be able to access any other new information we may choose to add to the context.  Passing the context (where users mostly ignore it) was a hedge against them wanting access to future APIs, or us wanting to add more data into the context in the future.  If that risk seems low, I could see the mental burden of a dummy argument outweighing the value of the hedge.  Your call.

RussTedrake, 2017-08-25:
Thanks.  I do think that {time,state,input,params} safely captures a huge class of the systems that we care about -- i feel there is little risk that we'll miss out on future context.

jwnimmer-tri, 2017-08-25:
Random idea: pass to the virtual methods (like `DoCalcVectorOutput`) a single argument instead of `double time` and some `VectorBlock`s, perhaps something like a `VectorContext vc` which is a sugar wrapper for the full `Context`.  The users do `vc.get_input()` and `vc.get_state()` or `vc.x()` etc. which provide vector blocks like we have now, but it (1) provides a seam to change the available information over time (e.g., add even better sugar, such as for `BasicVector` subtypes), with disturbing existing signatures and downstream code and (2) provides meaningful _names_ for the arguments.  I worry that once we have three `VectorBlock`s passed as arguments, there is some risk of getting them all jumbled up.

RussTedrake, 2017-08-25:
I like the suggestion; added flexibility but slightly higher cost for our "hello world" users.  Although I still think that part of the VectorSystem concept is to ignore all the other features for simplicity, so really am not worried (much) about future changes. 

@mposa and others -- would appreciate your perspective.  The question is, how much more palatable is 
`DoCalcTimeDerivatives(const double tim, const Vector& state, const Vector& input, const Vector& params, Vector* derivatives)`
vs  
`DoCalcTimeDerivatives(const VectorContext& vs, Vector* derivatives)`
proposed above?

jwnimmer-tri, 2017-08-25:
Another reason in favor of `VectorContext` formulation would be be lazy evaluation, e.g., don't evaluate the `input` to pass into `DoCalcTimeDerivatives(..., const Vector& input, ...)` when the system isn't direct feedthrough.  In the `DoCalcTimeDerivatives(const VectorContext& vc, Vector* derivatives)` form, only calling `vc.u()` would pull on the input.

RussTedrake, 2017-08-26:
ah.  that is a very good reason indeed

mposa, 2017-08-26:
My gut reaction is that VectorContext doesn't change a whole lot from standard Context, but the extra sugar would be (marginally) helpful. As to Russ' original suggestion, I don't know how much would have to change to move fully to (t,x,u). Is it possible to have `DoCalcTimeDerivatives(const double time, const Vector& state, const Vector& input, const Vector& params, Vector* derivatives)` just construct a Context and then pass that through to the default `DoCalcTimeDerivatives`?

RussTedrake, 2017-08-26:
@mposa -- we already have the implementation of what you suggest working, but currently provide
`DoCalcTimeDerivatives(const Context&, const Vector& input, const Vector& state, Vector* derivatives)`.  My suggestion was that it might be even more helpful / tutorial to move the few remaining things that people might need from the `Context` (time and parameters) into the list, and drop the rest of the `Context`.   

You're a perfect representative of the audience i'm trying to appeal to here, and it sounds like it IS more appealing to you than `VectorContext`?

mposa, 2017-08-26:
Ah I think I see. I think it would be useful to have context free
alternatives for some basic function calls (but preserving the full
functionality for advanced use).

RussTedrake, 2017-08-26:
maybe it's worth subclassing `VectorSystem` with the added simplicity.  something like `SimpleVectorSystem`.  I could even see dropping parameter support for that subclass.  

@jwnimmer-tri , @sherm1 -- fwiw, this discussion was inspired by my watching the reaction of potential users who are not c++ experts when they look at the tutorials (e.g. simple_continuous_time_system) and also by my wish to very rapidly prototype a variety of simple systems for unit testing the system optimization tools.

jwnimmer-tri, 2017-08-26:
More concrete example, from https://github.com/RobotLocomotion/drake/blob/master/drake/examples/simple_continuous_time_system.cc:

How it works on master now:
```c++
  // y = x
  virtual void DoCalcVectorOutput(
      const drake::systems::Context<double>& context,
      const Eigen::VectorBlock<const Eigen::VectorXd>& input,
      const Eigen::VectorBlock<const Eigen::VectorXd>& state,
      Eigen::VectorBlock<Eigen::VectorXd>* output) const {
    drake::unused(context, input);
    *output = state;
  }
};
```

Proposal 1:
```c++
  // y = x
  virtual void DoCalcVectorOutput(
      double time,
      const Eigen::VectorBlock<const Eigen::VectorXd>& state,
      const Eigen::VectorBlock<const Eigen::VectorXd>& input,
      const Eigen::VectorBlock<const Eigen::VectorXd>& params,
      Eigen::VectorBlock<Eigen::VectorXd>* output) const {
    drake::unused(time, input, params);
    *output = state;
  }
};
```

Proposal 2:
```c++
  // y = x
  virtual void DoCalcVectorOutput(
      const VectorContext& c,
      Eigen::VectorBlock<Eigen::VectorXd>* output) const {
    *output = c.state();    // alternatively, *output = c.x(); also works
  }
};
```

RussTedrake, 2019-03-25:
I think this has officially stalled.  With `LeafSystem` getting easier to use, and the existence now of `SymbolicVectorSystem` which can create simple systems in two lines.  Will close for now.

closed 2019-03-25 by RussTedrake