RobotLocomotion/drake #6918: valkyrie_simulation fails to load model in visualizer

hongkai-dai, 2017-08-25:
When @edrumwri and I launch `examples/valkyrie/valkyrie_simulation`, we see an error in the `drake_visualizer` as
```
failed to resolve filename, unknown package: package://Valkyrie/urdf/model/meshes/pelvis/pelvis.dae
warning, cannot find file: package://Valkyrie/urdf/model/meshes/pelvis/pelvis.dae
Traceback (most recent call last):
  File "/home/hongkai/drake-distro/build/install/lib/python2.7/dist-packages/director/lcmUtils.py", line 119, in handleMessage
    callback(msg)
  File "/home/hongkai/drake-distro/build/install/lib/python2.7/dist-packages/director/drakevisualizer.py", line 344, in onViewerLoadRobot
    self.addLinksFromLCM(msg)
  File "/home/hongkai/drake-distro/build/install/lib/python2.7/dist-packages/director/drakevisualizer.py", line 368, in addLinksFromLCM
    self.addLink(Link(link), link.robot_num, link.name)
  File "/home/hongkai/drake-distro/build/install/lib/python2.7/dist-packages/director/drakevisualizer.py", line 291, in __init__
    self.geometry.extend(Geometry.createGeometry(link.name + ' geometry data', g))
  File "/home/hongkai/drake-distro/build/install/lib/python2.7/dist-packages/director/drakevisualizer.py", line 264, in createGeometry
    polyDataList, visInfo = Geometry.createPolyDataFromFiles(geom)
ValueError: need more than 0 values to unpack

```

But I can find that `dae` file in `drake/examples/valkyrie/urdf/model/meshes/pelvis/pelvis.dae`. Does that mean `package://Valkyrie` does not map to `drake/examples/valkyrie` in the drake visualizer?

@patmarion

jwnimmer-tri, 2017-08-25:
Please post the exact commands you are running.

hongkai-dai, 2017-08-25:
We did
```
$ bazel test //drake/examples/valkyrie:valkyrie_simulation
```
in one terminal. On another terminal, we launch the `drake-visualizer` as
```
$ ./bazel-bin/tools/drake_visualizer &
```
Then on the terminal where we launch the `drake-visualizer`, we see the error message above.

jwnimmer-tri, 2017-08-25:
Other than `bazel test` vs `bazel run` (which likely does not make any difference), this seems like a duplicate of #6834.  There were several problems identified there; TTBOMK none of the fixes have been PRd yet.

hongkai-dai, 2017-08-25:
Thanks! I will close this issue, as it is a duplication of #6834

closed 2017-08-25 by hongkai-dai