RobotLocomotion/drake #6929: Move solver options into MathematicalProgramSolverInterface

hongkai-dai, 2017-08-28:
Currently `MathematicalProgram` stores the options for each solver, as a map from a string to a double/int/string
```C++
class MathematicalProgram {
  private:
    std::map<SolverId, std::map<std::string, double>> solver_options_double_;
    std::map<SolverId, std::map<std::string, int>> solver_options_int_;
    std::map<SolverId, std::map<std::string, std::string>> solver_options_str_;
}
```
Then inside each subclass of `MathematicalProgramSolverInterface`, it checks the options stored inside `MathematicalProgram`. This creates two issues

1. The user need to know which solver `MathematicalProgram` calls, and then set the option for that solver.
2. Some solvers do not access its options by a string. For example, in Mosek, all parameters are enums, rather than a string.
   ```C++
   enum MSKiparam_enum {
     MSK_IPAR_BEGIN                                 = 0,
     MSK_IPAR_END                                   = 185,

     MSK_IPAR_ANA_SOL_BASIS                         = 0,
     MSK_IPAR_ANA_SOL_PRINT_VIOLATED                = 1,
     MSK_IPAR_AUTO_SORT_A_BEFORE_OPT                = 2,
   }
   ```

I think the solution is to move these options inside each solver. So MosekSolver will have a data field
```C++
class MosekSolver {
  public:
    void SetSolverOptions(MSKiparam_enum, int);
  private:
    std::map<MSKiparam_enum, int> options_int_;
}
```
@EricCousineau-TRI

sammy-tri, 2017-08-28:
I can envision one case where this might not work (I'm not sure this case is worth supporting, but I'll throw it out there).  It's possible that some class might construct a MathematicalProgram without knowing what solver will be invoked (e.g. inverseKin not knowing if SNOPT or IPOPT will be called).  In this case it's possible that the calling code might want to choose reasonable default parameters for both solvers and set the MathematicalProgram up that way and let whichever solver is called run with the specified options.

Not that we actually do this, and after writing the above I'm even less sure it's worth it.  Are there other cases where some type of problem could use (for example) either Mosek or Gurobi and want to set options for both?

hongkai-dai, 2017-08-28:
@sammy-tri yes you are right, it is possible that we do not know which solver should be called, so we just set the options for all possible solvers.

The problem I run into is that Mosek does not access its parameters as strings. Instead it defines all the parameters by enum. We could keep the current interface, and add a `SetSolverOption` in `MosekSolver` class to handle this special case, as

```C++
class MosekSolver : public MathematicalProgramSolverInterface {
 public:
  void SetSolverOption(MSKiparam_enum, int);
 private:
  std::map<MSKiparam_enum, int> options_int_;
}
```

EricCousineau-TRI, 2017-08-30:
If we wanted to generalize (and potentially join at some point) solvers options, should we make our own string mappings that map to the given solver-specific enums, to keep it all within `MathematicalProgramSolverInterface::SetSolverOption`?

(How does JMP handle this?)

rdeits, 2017-08-30:
JuMP puts all options inside the `Solver` itself and makes no attempt to have cross-solver options. I think that's pretty reasonable, since every solver has a wildly different set of options that can have very different effects on the actual solution time and quality. Even something as simple as `verbosity` means different things to different solvers, since many of them have a variety of levels of output they can produce.

jwnimmer-tri, 2017-08-30:
FWIW Back when I was going to be working on this, the design I had in mind was for `MathematicalProgram` offer a different `SolverOptionSet` for each different `SolverId` (i.e., a `map<SolverId, SolverOptionSet>`, and then also for the `FooSolver` to let you directly customize its `SolverOptionSet`.  During automated dispatch (or earlier?), the `SolverOptionSet` would be propagated from the `MathematicalProgram` to the `FooSolver`.

Then `SolverOptionSet` just has to be able to reasonably denote all of the ways one might want to customize.  My plan for its API was `string` as a key and `AbstractValue` as the value.  Most keys are strings already; keys that are, say, enums would support either the string value of the backing integer, or a string name of the enum.  (The solver would check for both keys.)  Values can they be any (copyable) type, which should cover all cases.

hongkai-dai, 2017-08-30:
@jwnimmer-tri I think `SolverOptionSet` makes sense, but I am unsure about using `string` as a key. Mosek has many parameters http://docs.mosek.com/7.1/toolbox/Parameters.html. It would be tedious to convert every string to the enum parameters. Moreover, the set of parameters change between each version of the solver.

jwnimmer-tri, 2017-08-30:
The `key` API could offer `int`-taking methods, which it converted to `string` under the hood, though?  It would also take only an hour to write a tool that scraped the enums and generated C++ list of strings, or somesuch.

jwnimmer-tri, 2019-01-16:
@hongkai-dai did #9904 sufficiently address this desire, such that we can close this issue?  Or do we need to add more methods to `class SolverOptions` for some use cases?  (If so, please elaborate.)

hongkai-dai, 2019-01-16:
Yes, I think now `SolverOptions` class is sufficient.

closed 2019-01-16 by hongkai-dai