RobotLocomotion/drake #6932: LeafSystem continuous state allocation vs models

SeanCurtis-TRI, 2017-08-28:
PROBLEM
========
There is a problem in the `LeafSystem`'s use of its model values.

- `model_continuous_state_vector_` is cloned in `LeafSystem::AllocateContinousState()` (if it exists).
- `LeafSystem::SetDefaultState()` sets all continuous (and discrete) state to the zero vector.

Essentially, values placed in the model vector are rendered meaningless if `SetDefaultState()` gets invoked. This seems like the exact *opposite* of what one would expect. By providing a model, the default state should not just get the dimension of the state vector from the model, but also the values.

Furthermore, the `SetDefaultState()` method only sets default values for continuous and discrete state. It doesn't touch abstract state at all.

PROPOSAL
=========
Allocate should be primarily responsible to extract the *structure* of the model (e.g., dimension of the state vector) as far as possible, and setting the default should apply the values from the models to the structure. Zeroing out vectors should only happen in the *absence* of model vectors.

sherm1, 2017-08-28:
:+1: Since AllocateContinuousState() is used for both state and state derivatives, it should stick to allocating and *not* set default values. It does not make logical sense to initialize derivatives with the default values of states. Better just to allocate and set to zero or (likely better) NaN. Then SetDefaultState() should copy the model values in.

RussTedrake, 2017-08-29:
FWIW -- i think that the primary role of passing in the model vector is to pass the runtime type.  (e.g. SimpleCarState is more than a BasicVector).  the values were not the point.  but if we can make the SetDefaultState more intuitive, then bonus.

jwnimmer-tri, 2017-08-29:
Yes indeed, though the secondary purpose was to conveniently provide the default values as well, as part of the same package.  If someone wants to change the `LeafSystem` implementation to overwrite the elements with `NaN`s during `Allocate` and then re-assign the model values during `SetDefault`, that's a-ok by me.

jwnimmer-tri, 2021-11-11:
Fixed in #7148.

closed 2021-11-11 by jwnimmer-tri