RobotLocomotion/drake #6943: Tiny change to simulation causes monolithic pick and place (scenario 1) to fail.

edrumwri, 2017-08-30:
As observed in PR #6912, a tiny change to the simulation causes the monolithic pick and place (scenario 1) to fail with a timeout due to:

[2017-08-29 15:21:11.216] [console] [info] kApproachPickPregrasp at 0.53
[2017-08-29 15:22:10.648] [console] [info] kApproachPick at 2.59
[2017-08-29 15:23:09.016] [console] [info] kGrasp at 4.61
[2017-08-29 15:23:11.656] [console] [info] Generated fixed plan at  0.269333    2.0944 -0.567478   1.23901 -0.599252   1.02662  0.900756
[2017-08-29 15:23:27.973] [console] [info] kLiftFromPick at 5.13
[2017-08-29 15:24:39.088] [console] [info] kApproachPlacePregrasp at 7.18
Restoration phase is called at point that is almost feasible,
  with constraint violation 2.775558e-16. Abort.
[2017-08-29 15:25:52.150] [console] [warning] IK FAILED at max constraint relaxing iter: 11
Restoration phase is called at point that is almost feasible,
  with constraint violation 5.551115e-16. Abort.
[2017-08-29 15:25:54.692] [console] [warning] IK FAILED at max constraint relaxing iter: 11

The description of the aforementioned tiny change to the simulation follows: the surface normal used in the compliant contact model is normalized before being used to compute a basis. This surface normal is *already* effectively unit length: normalization causes the length of the vector to change by no more than 1e-15. _This less-than 1e-15 change in the normal (per ODE evaluation) is sufficient to cause the state to evolve to a (nearby) configuration where the IK solver fails._ 

To reproduce this issue, change "const Vector3<T> this_normal = pair.normal;" on Line 109 of drake/multibody/rigid_body_plant/compliant_contact_model.cc to "const Vector3<T> this_normal = pair.normal / pair.normal.norm();"

naveenoid, 2017-08-30:
Transferring to @avalenzu since he has been looking into IK performance in the pick and place scenarios.

sherm1, 2017-08-30:
There is something very odd about this failure:
> Restoration phase is called at point that is almost feasible,
with constraint violation 2.775558e-16. Abort.

It is hard to imagine any numerical definition of feasibility for which a violation of 2e-16 would not be acceptable!

edrumwri, 2017-08-30:
Good point- I'll be curious to hear whether the resolution is simply modifying that tolerance.

avalenzu, 2017-08-31:
> Restoration phase is called at point that is almost feasible, with constraint violation 2.775558e-16. Abort.

Did you build and execute `pick_and_place_scenario_1` without setting `--config snopt`? T he error message above from Ipopt. This test should only be running run when `--config snopt` is set (i.e. `bazel test` would not run this without `--config snopt`.  I'll see if I can reproduce the CI failure locally with snopt.

avalenzu, 2017-08-31:
I've been able to reproduce the CI failure locally - no IK failures, the arm just rams into one of the posts and the sim explodes. I have a version of the monolithic demo with re-vamped planning that is (qualitatively) more robust, but I havn't PR'ed it yet - still testing on the real robot.

@sammy-tri , @naveenoid , @siyuanfeng-tri , @RussTedrake : The current monolithic pick and place demo seems too brittle to be an effective regression test in CI - too many false positives. Should we consider disabling it until we have a more robust version?

naveenoid, 2017-08-31:
I think that atleast until your PR lands we need to move it out of CI as you recommend.

sammy-tri, 2017-09-01:
1.  I would still like to hear if this is ipopt instead of snopt and what happens with snopt.
1.  I would expect virtually any simulation to fail with the way things are now if you bashed the arm into fixed geometry.
1. Which type of 'brittle'?   I think the brittleness in the planner (sensitive dependence on initial conditions) might be an issue which could rise to removing the test from CI (or making failures from certain scenarios non-fatal).  I think any brittleness resulting from simulation issues (such as failures because the placed object is still moving around after release and tripping the velocity limits for the final conditions) indicates that this test is contributing value in CI.

edrumwri, 2017-09-01:
I actually could not run it using the same settings that was used in CI, so
I ran the test locally on my machine and then observed a failure that
appeared consistent with what I was seeing in the CI output.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

avalenzu, 2017-09-05:
@sammy-tri - my comment about being able to reproduce the failure locally referred to running the test with snopt. 

My diagnosis is brittleness in the planner - small changes to the initial condition led to a markedly different plan, which in turn led to colliding with the fixed geometry.

edrumwri, 2017-09-06:
Is there agreement on the proposed resolution of this issue (disabling the monolithic p&p example in CI)?

sammy-tri, 2017-09-11:
My vote would be to disable only scenario 1 for now...  though I'm OK with disabling them all if it comes to that, there should be significant improvements to the pick and place planner coming fairly soon.

edrumwri, 2017-09-13:
Ok, shall I make this change?

sammy-tri, 2017-09-13:
If it's necessary to unblock your other work (and it sounds like it is), then I say do it.

SeanCurtis-TRI, 2017-09-15:
FTR What is the mechanism for "disabling" a test?

soonho-tri, 2017-09-15:
@SeanCurtis-TRI , you can do it by adding `DISABLED_` in a test name. For example,

```c++
GTEST_TEST(PolynomialTest, DISABLED_Simplification) {
```

soonho-tri, 2017-09-15:
or.. do you mean disabling a test at the bazel level?

SeanCurtis-TRI, 2017-09-15:
In this case, the test isn't executed in gtest. So, that wouldn't work. So, I had hoped for a bazel-level mechanism that means that the build target can be executed, but isn't done so by CI.

sammy-tri, 2017-09-15:
Adding `--quick` to the pick and place test args should effectively disable the test (along with a TODO to remove the argument).  A human could still run it by hand by removing that argument.

I don't know how to do this with bazel.  The pick and place test gained the `--quick` argument in the first place because I wanted to conditionally disable the tests in debug builds (see `pick_and_place_test.bzl`).

sammy-tri, 2017-09-15:
Or I suppose just delete the rule from the BUILD file and leave a note regarding the configuration which should eventually be restored when the issue is resolved.

soonho-tri, 2017-09-15:
One way we can try is to introduce and add `ci_only` tag and modify https://github.com/RobotLocomotion/drake/blob/master/tools/bazel.rc#L19 to have `-ci_only` so that by default it's not triggered. We will need an update in CI script to enable those tests with `ci_only` tag.

jwnimmer-tri, 2017-09-15:
That sounds complicated.

Can someone summarize what exactly is wanted?  If you just want to pass `--quick` to scenario 1, why not just add that to the `args` already in the `BUILD` file?
```
pick_and_place_test(
    name = "pick_and_place_scenario_1",
    # TODO(sam.creasey) This should be target 1, but that requries
    # offsetting the gripper location to deal with the larger box,
    # which isn't currently supported.
    args = [
        "--quick",  # TODO() XXX This is failing, remove me.
        "--target=2",
        "--orientation=1.57",
        "--start-position=0",
        "--end-position=2",
        # TODO(sam.creasey) This is an artisanally crafted timestep
        # intended to dodge sensitive dependence on initial conditions
        # in the IK planner.  Depending on slight changes in the arm's
        # configration after lifting the item from the pick, the IK
        # planner (particularly when running under SNOPT) can produce
        # radically different results.  This value produces usable
        # (but different!) results on Mac vs. Linux builds.  See #6811
        # for more information.
        "--dt 6e-4",
    ],
```

SeanCurtis-TRI, 2017-09-15:
@jwnimmer-tri I think you've lost the goal in the prolonged discussion of the means.

The goal is to "disable" a test in CI. The PR in its current state simply nukes the build target. That felt a bit heavy handed. So, what are the alternatives?

What you cite is one suggestion for simulating removing it from CI (i.e., make its execution trivially successful). That seems the best option available given current state of affairs.

jwnimmer-tri, 2017-09-15:
Ah.  So `tags = ["manual"],` then.

SeanCurtis-TRI, 2017-09-15:
It even *sounds* right.  Thanks, @jwnimmer-tri

edrumwri, 2017-09-15:
Ok, thank you for the elegant solution to the problem, Jeremy. PR has been merged and the monolithic P&P Scenario 1 disabled from running in CI.

closed 2017-09-15 by edrumwri