RobotLocomotion/drake #6944: Eigen assert on AutoDiffXd expressions with >2 operands and some empty derivatives

jwnimmer-tri, 2017-08-31:
This problem was identified in #6940.

```c++
GTEST_TEST(EigenTest, ExpressionTest) {
  using T = AutoDiffd<-1>;
  T x{1.0};
  T y{1.0};
  T z{1.0};
  z.derivatives() = Eigen::VectorXd::Ones(2);

  x * y * z;
}
```

The `-1` above means `Xd`.  In debug builds, this aborts with an internal Eigen assertion.  Changing all three to use empty derivatives success.  Changing the `-1` to a `2` succeeds.

The fault appears to be related to `make_coherent` not promoting operands to have matching derivatives.

I will post a more thorough, standalone description here later.

jwnimmer-tri, 2017-08-31:
Reading Eigen code, it looks like `AutoDiffScalar` tries to `internal::make_coherent` to promote empty derivatives up to match the other operand in many cases

The problem is in
```c++
class AutoDiffScalar
    operator*(const AutoDiffScalar<OtherDerType>& other) const
    {
      internal::make_coherent(m_derivatives, other.derivatives());
      return MakeAutoDiffScalar(
        m_value * other.value(),
        (m_derivatives * other.value()) + (other.derivatives() * m_value));
    }
```

on the line
```c++
(m_derivatives * other.value()) + (other.derivatives() * m_value))
```
with the failing assertion being `Assertion 'aLhs.rows() == aRhs.rows() && aLhs.cols() == aRhs.cols()' failed.`

The type of that expression is
```c++
Eigen::CwiseBinaryOp<
  Eigen::internal::scalar_sum_op<double, double>,
  const Eigen::CwiseBinaryOp<
    Eigen::internal::scalar_product_op<double, double>,
    const Eigen::CwiseBinaryOp<
      Eigen::internal::scalar_sum_op<double, double>,
      const Eigen::CwiseBinaryOp<
        Eigen::internal::scalar_product_op<double, double>,
        const Eigen::VectorXd,
        const Eigen::CwiseNullaryOp<
          Eigen::internal::scalar_constant_op<double>,
          const Eigen::VectorXd
        >
      >,
      const Eigen::CwiseBinaryOp<
        Eigen::internal::scalar_product_op<double, double>,
        const Eigen::VectorXd,
        const Eigen::CwiseNullaryOp<
          Eigen::internal::scalar_constant_op<double>,
          const Eigen::VectorXd
        >
      >
    >,
    const Eigen::CwiseNullaryOp<
      Eigen::internal::scalar_constant_op<double>,
      const Eigen::VectorXd
    >
  >,
  const Eigen::CwiseBinaryOp<
    Eigen::internal::scalar_product_op<double, double>,
    const Eigen::VectorXd,
    const Eigen::CwiseNullaryOp<
      Eigen::internal::scalar_constant_op<double>,
      const Eigen::VectorXd
    >
  >
>::CwiseBinaryOp(const Lhs &, const Rhs &, const BinaryOp &)
```
The coherency promotion signature is:
```
struct make_coherent_impl<
  Matrix<A_Scalar, A_Rows, A_Cols, A_Options, A_MaxRows, A_MaxCols>,
  Matrix<B_Scalar, B_Rows, B_Cols, B_Options, B_MaxRows, B_MaxCols> > {
```

My guess is that the Cwise intermediate results are not matching the Matrix specialization of `make_coherent_impl`, but rather hitting the no-op default implementation of it.

It is possible that https://bitbucket.org/eigen/eigen/diff/unsupported/Eigen/src/AutoDiff/AutoDiffScalar.h?diff2=815734e048fd&at=default was what used to make this work? It was removed around a year ago.

My best guess at a fix would be to further specialize `make_coherent_internal` to handle Cwise ops, and/or possibly restore those removed Cwise specializations that forced MatrixXd intermediate results, and/or some other alteration to how Eigen types get forced in order for `make_coherent` to work. Some `drake/common` unit tests for all of this would also be in order.

jwnimmer-tri, 2017-08-31:
So here's the deal.  The way `make_coherent` work is that it takes const references to the two ADS operands.  Then, through overloading, whichever of them are `Matrix` it **casts away the const**, checks if they are `Eigen::Dynamic`, and if so resizes them to match the other operand.  So if both are dynamic and empty, they will remain empty; if one is non-empty then it will promote the other to non-empty.

The problem comes when one or more of the operands are expression trees (`CwiseBinaryOp`), etc.  Those cannot be resized, so if you are unlucky and operands there are both empty, then the subexpression will remain empty, and then trip the assert.

I put together a patch so that `make_coherent` descends through `CwiseBinaryOp` and recursively promotes all the `Eigen::Dynamic` it finds.  That worked as far as it went.  The problem is that there are often `CwiseNullaryOp` terms as well (e.g., element-wise multiplying a 1d times an Xd), and I cannot resize those.  So that was a dead-end.

The solution that actually works is to change the overload selection for operators on `ADS<VectorXd>`.  If we have e.g. `ADS<Vector2d>` then the overload stays as-is.  If all operands are `ADS<VectorXd>`, then instead of returning an expression tree, we immediately evaluate the term and return a `ADS<VectorXd>` by value, not a promise.

That looks like this:
```c++
namespace Eigen {
const AutoDiffScalar<Eigen::Matrix<double, Eigen::Dynamic, 1>>
operator*(
    const AutoDiffScalar<Eigen::Matrix<double, Dynamic, 1>> a,
    const AutoDiffScalar<Eigen::Matrix<double, Dynamic, 1>>& b)
{
  internal::make_coherent(a.derivatives(), b.derivatives());
  return MakeAutoDiffScalar(
      a.value() * b.value(),
      (a.derivatives() * b.value()) + (b.derivatives() * a.value()));
}
} // namespace Eigen
```

To use Xd successfully, we will have to overload all operators that end up using Cwise at all to have an overload like this.  I think that is only add, sub, and mul (and possibly their op= mates).

jwnimmer-tri, 2017-08-31:
@soonho-tri Would you be interesting in taking over this issue and putting together the `eigen_autodiff_overloads.h` patch and unit tests to solve it, and possibly report upstream?

soonho-tri, 2017-08-31:
@jwnimmer-tri , sure. I'll take care of it. Thanks for all the investigation!

soonho-tri, 2017-08-31:
@jwnimmer-tri , please feel free to un-assign yourself.

soonho-tri, 2017-09-05:
FTR, I had f2f session with @jwnimmer-tri and found that we need to provide more than the three operations mentioned above (+, -, /). It seems that we need to provide overloading for all unary/binary operations.

soonho-tri, 2017-09-07:
@jwnimmer-tri , I have a [WIP branch](https://github.com/soonho-tri/drake/tree/pr-fix-autodiffxd).
In this branch, the existing tests passed without the "resize-and-setzeros" hack.

Currently, I'm repeating what Eigen's `AutoDiffScalar.h` did. PTAL https://github.com/soonho-tri/drake/blob/pr-fix-autodiffxd/drake/common/autodiff_overloads.h#L129-L137. In the last f2f, we talked about calling Eigen's implementation in our specialized functions, but when I try to implement that, I couldn't find a way to call them. Do you have a good solution for that?

soonho-tri, 2017-09-07:
FTR, @jwnimmer-tri suggested two options -- 1) using `Eigen::Ref`, 2) copying `drake::AutoDiffXd` to a variant of `AutoDiffXd` with `Eigen::DontAlign`.

@jadecastro , could you point me a couple of lines where we use the workaround for AutoDiffXd? I'd like to remove them and see if this fix is working before I open a PR.

jadecastro, 2017-09-07:
@soonho-tri The places where I use that hack is actually in a (stale) PR in-flight. This is the place in that PR where I use it:
https://github.com/RobotLocomotion/drake/pull/6614/files#diff-e5fe184ed2f7c194348eecc1a74b3b29R30

You're welcome to work off of that PR if it helps to see if it works and then coordinate from there.

closed 2017-09-18 by soonho-tri

jwnimmer-tri, 2017-09-18:
Is there any upstream bug report or discussion forum we could link to here?  This seems like the kind of thing that should eventually be fixed upstream, too.

soonho-tri, 2017-09-18:
@jwnimmer-tri , I'll file a report and put a link in this thread.

jwnimmer-tri, 2018-08-03:
@soonho-tri Could you pose the upstream bug report link here, please?

soonho-tri, 2018-08-03:
@jwnimmer-tri , I realized that I didn't file the report. I'll do it this time for sure. I've contacted the Eigen team to have an account for their Bugzilla system. Also checked that the bug still exists in the latest version (`3.3.5`).

soonho-tri, 2018-08-03:
Here is the bug report: http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1582

PR #7007 (2017-09-18): Provide overloading for Eigen's AutoDiffXd
which evaluates terms immediately and returns an AutoDiffXd instead of expression trees.

1. Provide a template specialization, `AutoDiffScalar<VectorXd>`, which modifies the return type of operators (`+`, `-`, `*`, `/`). 
2. Provide overloading of math functions (such as `sin`, `cos`, `pow`, `log`, `exp`, ...) for `AutoDiffXd`.

A consequence of having our own version of `AutoDiffScalar<VectorXd>` is that we need to provide its definition before any code instantiates it. To make sure that, I've replaced `#include <unsupported/Eigen/AutoDiff>` with `# [... cut]