RobotLocomotion/drake #6949: Excessive warning printout: "DrakeVisualizer::Publish() called before PublishLoadRobot()"

samzapo, 2017-08-31:
From:
https://github.com/RobotLocomotion/drake/blob/f1656ce15b9151dc709c9c0f309fd11ef2bf9198/drake/multibody/rigid_body_plant/drake_visualizer.cc#L143

Example:
`[2017-08-31 16:50:40.870] [console] [warning] DrakeVisualizer::Publish() called before PublishLoadRobot()`

printed at ~20Hz for a working robot with successful visualization.

SeanCurtis-TRI, 2017-09-01:
Thanks. I'll take a look.

samzapo, 2017-09-01:
Occurs until the simulator is able to complete its first step.
DrakeVisualizer information is init'ed after the first call to DrakeVisualizer::DoCalcDiscreteVariableUpdates(.) which calls PublishLoadRobot()

SeanCurtis-TRI, 2017-10-06:
#7207 Provides a band-aid. It doesn't "fix" it. But it makes this outcome less likely.

jwnimmer-tri, 2019-04-24:
@SeanCurtis-TRI to the best of my knowledge, this doesn't happen with MultibodyPlant, only RigidBodyPlant?  If so, we should close this out as "won't fix".

SeanCurtis-TRI, 2019-04-25:
Agreed.

closed 2019-04-25 by SeanCurtis-TRI