RobotLocomotion/drake #6951: LeafSystem::DoHasDirectFeedthrough is weird API for subclasses to override

jwnimmer-tri, 2017-09-01:
`LeafSystem` provides a method `DoHasDirectFeedthrough(const SystemSymbolicInspector* sparsity, int input_port, int output_port)` for subclasses to override, for cases when the symbolic form is either unavailable, or overkill for this purpose.

First, it is just plain weird for user subclasses to mention `SystemSymbolicInspector` when manually overriding sparsity.  I believe every current use does not touch this parameter.

Second, it means that even if a user subclass can directly and efficiently express the sparsity, the framework goes to all of the trouble to (try to) prepare the `SymbolicSystemInspector` (which might be extremely expensive), only to throw it away.

We should adjust `LeafSystem`'s protected interface toward subclasses to be more elegant and efficient.

I will post a rough proposal here once I have one.

/CC @RussTedrake

jwnimmer-tri, 2017-09-01:
LeafSystem could offer a few kinds of things to subclasses:

(1) For some (or likely all) of the `Has(Any)DirectFeedthrough` methods, offer a protected `Set(Any)DirectFeedthrough` where the subclass can directly express the feedthrough, e.g., during construction time.  Only if these have not been set would the symbolic form be attempted.  This has the benefit of concise flexibility -- the any / per output / per input-output pair would all be directly expressible.

(2) For some (or all) of the `DeclareVectorOutputPort` methods, add a parameter to express that the output has no direct feed-through.  This would put the onus at the point of declaration, which is perhaps more cohesive and likely to be correct.  The downside is it makes the declaration methods even more complicated.

(3) Alter the `DoHasDirectFeedthrough` signature that is being overriden.  Remove the inspector as a  argument, and of instead of returning a yes/no bool, return a yes/no/unknown enum, where only if we see "unknown" (the default) would LeafSystem attempt to use symbolic form.

sherm1, 2017-09-01:
> For some (or all) of the DeclareVectorOutputPort methods, add a parameter to express that the output has no direct feed-through.

These methods will be getting a dependencies list as part of the caching implementation. That should serve this purpose too.

jwnimmer-tri, 2017-09-01:
@sherm1 Great!  In that case, do you have a sense of whether (1) or (3) or (something else) would be most compatible with that approach?  It seems likely that I will PR some intermediate improvement in the meantime.

SeanCurtis-TRI, 2017-09-01:
I almost wonder if it's worth it. The dependencies inherent in @sherm1's caching system should supplant our current mechanism for detecting algebraic loops. Almost. It does depend on the author of a `System` *correctly* articulating dependencies. To serve completely as a mechanism for algebraic loops we would have to:
  a) Have some form of validation (that the dependencies list are indeed correct), or
  b) a system that precludes accessing resources that haven't been declared as dependencies.

(Or maybe some alternative solution that doesn't come to me right now.) Otherwise, we end up with two mechanisms that should mirror each other if things are correct and deviate if things are broken.

jwnimmer-tri, 2017-09-01:
It is table stakes for sherm1's dependency declarations to be programmatically checked or enforced at runtime with something like (b).  It should not hit master without that.  I agree that such a system would (in all or almost all cases) supplant my more limited proposals above, and that we should deprecate and remove the more limited APIs after the richer, controlling one is available.

However, this fixup was a specific request from Russ that I want to merge in a week or two, and not wait for caching to land first.  If the port-declaration-time dependency APIs (+ enforcement) could merge in the next week or two, that would be a fine option also.

jwnimmer-tri, 2017-09-01:
So as a stop-gap, it seems like (3) would be the smallest change that removes the SystemInspector when unwanted, but doesn't add any new APIs that we have to roll back later -- it just changes the signature of a (lightly used) API.  I think its my preference, unless sherm1 thinks that a different API choice would meld better into his future design.

sherm1, 2017-09-01:
> So as a stop-gap, it seems like (3) would be the smallest change ...

That would be fine with me, Jeremy. 

> It should not hit master without [automatic validation of dependencies].

FYI, my current intention is *not* to include that in the first batch of PRs. There is a huge amount of infrastructure to review to ensure flawless & performant functioning of the cache assuming the dependencies are given. Validating that those dependencies are correct is an additional project. But we can discuss further with code on the table if you want.

RussTedrake, 2017-09-02:
Sorry for being late to the party.  I want to add that HasDirectFeedthrough is the first example of a pattern that we will likely need to repeat for other system properties.  IsTimeVarying is the other example that I already consider to be in the queue -- again we'd like for a system to be able to inspect this property automagically (I just added it to the SystemSymbolicInspector), but also allow for the leaf system to declare this so that we can 
1. avoid the cost of producing a symbolic form for big systems, and 
2. set the attribute for systems that do not or cannot implement DoToSymbolic.

PR #6973 (2017-09-06): Remove symbolic inspector from LeafSystem feedthrough API
`LeafSystem::DoHasDirectFeedthrough` previously accepted an already-constructed `SystemSymbolicInspector`, which means that we were trying to create symbolic form even for systems that directly expressed their sparsity.  It was also a confusing API for users to understand.

Fixes #6951.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6973)
closed 2017-09-06 by jwnimmer-tri