RobotLocomotion/drake #6959: Symbolic Expression support for random numbers

RussTedrake, 2017-09-03:
Related to #6374... I believe my first requirement is to be able to represent some randomness in symbolic expressions.  I believe that I want to add two new types to symbolic::ExpressionKind 
1) `random_uniform_real`
and 
2) `random_normal`
(following the naming conventions in http://en.cppreference.com/w/cpp/concept/RandomNumberDistribution )
Unless there is something that I am not thinking of, these should act like constants in every symbolic operator overload.  @soonho-tri -- feel free to implement it however is best (special values for ExpressionKind::Constant, or whatever).

Following #6374, I believe that if this is implemented, then i can obtain the process and sensor noise by overloading `ToAutoDiffXd` in the `RandomSource<std::uniform_real_distribution<double>>` and `RandomSource<std::normal_distribution<double>>` systems to return a `ConstantSource<Expression>` system with the values above.  Drake gurus -- please check me on that.

RussTedrake, 2017-09-03:
For completeness, if we can implement `drake::symbolic::uniform_real_distribution<symbolic::Expression>` (and normal_distribution) that overloads `operator()` to always return the `random_uniform_real`, etc, then I think all of the use cases I have in mind can be pretty elegant.

soonho-tri, 2017-09-03:
I'll work on it.

The introduction of randomness in symbolic expressions affects how we evaluate them. To evaluate a symbolic random-value, we need a way to pass an RNG (random number generator) to it.

One approach is to extend the current member method
```c++
double Expression::Evaluate (const Environment &env) const
```
to take an RNG as an argument. It might be still good to keep the current version and throws an exception if it hits a random kind.

Another approach is to switch to a separate class for symbolic evaluation, i.e. `symbolic::Evaluator`. It can hold an RNG as a member (and Environment as well).

As I implement things, I'll check the pros/cons of each approach. @RussTedrake & @jwnimmer-tri, please leave your inputs here.

RussTedrake, 2017-09-03:
Good point.  I think a first pass would still be useful if Evaluate were to throw if it is called on an Expression containing a random element.  But I agree this is something to resolve in the long run.

jwnimmer-tri, 2017-09-03:
Agreed on "throw" for during Evaluate for now.

Longer term, I would lean toward explicitly having to substitute in randomness, akin to substituting in variable values.  The difference would be each random term takes on a new value, distinct from the other terms, as if they were distinct variables.  Whether that is through the substitution APIs directly, or a Evaluator wrapper, I think would let users to choose which they prefer.

RussTedrake, 2018-12-17:
First thoughts on this appearing inhttps://github.com/RobotLocomotion/drake/compare/master...RussTedrake:symbolic_rand?expand=1

soonho-tri, 2018-12-17:
@RussTedrake , could you design-review my WIP branch?

https://github.com/RobotLocomotion/drake/compare/master...soonho-tri:symbolic-random-variables?expand=1

 - Instead of extending the `symbolic::Expression`, I've introduced new "random variable" types (`RANDOM_UNIFORM`, `RANDOM_GAUSSIAN`, `RANDOM_EXPONENTIAL`) in `symbolic::Variable` class.
 - I provided a specialization of `std::uniform_real_distribution` for `symbolic::Expression`. Please check the usage in the test code and let me know if this is what you want.
 - I've added two `Evaluate` functions in a separate header `symbolic_expression_evaluator.h`:

    ```c++
    template <typename T>
    double Evaluate(const Expression& e, const Environment& env, T* generator);
    
    template <typename T>
    double Evaluate(const Expression& e, T* generator);
    ```

soonho-tri, 2018-12-17:
- I wanted to check if you really needed `is_uniform_real(...)` in your original test code. In my version, we have random variables, not expressions. It's easy to check if one variable is uniform-real or not. However, it is not immediately straightforward if an expression is uniform-real (though we can still check if it an expression is a linear combination of uniform-real variables and constants).

RussTedrake, 2018-12-17:
don't need is_uniform_real explicitly (not yet at least!)

RussTedrake, 2018-12-17:
Just reviewed your WIP.  Looks fantastic!  Thank you!
N.B. -- there are more methods of the uniform_real_distribution that we will want to implement at some point to flesh out support (to mimic std::uniform_real_distribution<double>).  Not sure if we should do them now or wait for later?

jwnimmer-tri, 2018-12-17:
Note that `std::uniform_real_distribution` uses the range `[a, b)` but drake's definition of kUniform is `[0, 1]`.  You will have to be careful when going back and forth between the two, as to whether the upper bound is an open or closed interval.

RussTedrake, 2018-12-17:
that sounds like a bug in drake's documentation (on me), as the implementation is always pulling from std::uniform_real_distribution

jwnimmer-tri, 2018-12-17:
I'm fine if we want to change drake docs.  It just means that the variable's mean will be 0.4999999999etc instead of 0.5; I don't know if that matters for our applications or not.

soonho-tri, 2018-12-17:
@RussTedrake , thanks for the review!

>  N.B. -- there are more methods of the uniform_real_distribution that we will want to implement at some point to flesh out support (to mimic std::uniform_real_distribution). Not sure if we should do them now or wait for later?

I'll check and try to implement as many as possible this evening. We can discuss and add more later, I belive.

Some caveats:

 - I've added a non-method function `Evaluate(e, env, &generator)` instead of adding `Expression::Evaluate(env, &generator)`. This is due to the fact that we need to introduce a template function (because of a generic-typed `generator`) and it should be in a header file as a result. It turns out that there could be a circular dependency between `symbolic_expression_visitor.h` and `symbolic_expression.h` which makes it complicated to add a method version.
 - Also note that `Expression::Evaluate` and `Evaluate(e, env, &generator)` essentially do the same thing.
 - For now, my plan is to keep the existing `Expression::Evaluate(env)` but refactor it to call the newly introduced function version, `Evaluate(e, env, nullptr)`. I'm not sure that if we want to deprecate `Expression::Evaluate(env)` though.

RussTedrake, 2018-12-17:
In the systems framework, I've typedef'd the std:mtl## to RandomGenerator.  I think it's unlikely that we'll need to use another.  i'd be ok using that type instead of a template type if that simplifies things

soonho-tri, 2018-12-17:
I see. Then, I'll add the non-template method version in the upcoming PR.

soonho-tri, 2018-12-18:
>  i'd be ok using that type instead of a Generator template type if that simplifies things

I've updated my WIP branch: https://github.com/RobotLocomotion/drake/compare/master...soonho-tri:symbolic-random-variables?expand=1

 - I've added `using RandomGenerator = std::mt19937;` in `class Expression`.
 - I've extended `Expression::Evaluate` to be:
    ```c++
       double Evaluate(const Environment& env = Environment{},
                       RandomGenerator* random_generator = nullptr) const;
    ```

TODO:
- [ ] Specialize `std::exponential_distribution` and `std::normal_distribution`.
- [ ] Add more tests.

RussTedrake, 2018-12-18:
Horrah!  A few more thoughts:

Some of the random distribution ideas are already in drake/common/constants.h.  I recommend moving the definition of RandomGenerator from system.h to somewhere in drake/common and having symbolic_expression using the same definition.

I think jeremy is right, and we should write [0,1) for the RANDOM_UNIFORM doxygen string

could probably use a little documentation in symbolic_variable for Variable::Type to make it more clear that the substitution of those variables is done automatically on Evaluate.

this is really beautiful.  what a nice addition!  please mark me as a feature reviewer when you're ready; I've been effectively feature reviewing as we go.

RussTedrake, 2018-12-18:
one more for my wishlist (not necessarily for the first PR): 
- [ ] Add python bindings

soonho-tri, 2018-12-18:
> Some of the random distribution ideas are already in drake/common/constants.h. I recommend moving the definition of RandomGenerator from system.h to somewhere in drake/common and having symbolic_expression using the same definition.

Done (in my WIP)

BTW, we're using Gaussian (e.g. `kGaussian`, `RANDOM_GAUSSIAN`) while we provides a specialization for `std::normal_distribution` (not, `gaussian_distribution`). Do you think it makes sense to do `s/Gaussian/Normal/` in our codebase?

RussTedrake, 2018-12-18:
I felt it was more self-sufficient with Gaussain -- and at least there is not some other `gaussian_distribution`), but I'd be aok with `s/Gaussian/Normal/`.  It would be an API break, but better now (where my code is probably the only code that uses it).

closed 2019-01-07 by RussTedrake