RobotLocomotion/drake #6960: Compute taylor expansion for symbolic expression

hongkai-dai, 2017-09-03:
When verifying stability using Sum of Squares, we need to approximate the dynamics as polynomials, using taylor expansion. We should consider that for a given symbolic expression, take the taylor expansion up to `N`th order. (Most likely N <= 3).

@soonho-tri @RussTedrake

RussTedrake, 2017-09-03:
The obvious alternative is to use recursive autodiff.  Do we think that we'll ever want to do autodiff for systems that we can't support with symbolic?

soonho-tri, 2017-09-03:
@hongkai-dai , FYI, implementing it shouldn't be difficult thanks to what we already have -- symbolic polynomial and `Differentiate` method.

hongkai-dai, 2018-09-25:
@soonho-tri added the feature already.

closed 2018-09-25 by hongkai-dai