RobotLocomotion/drake #6961: Clean up system/context safe vector down casting

RussTedrake, 2017-09-04:
Off the top of my head:
Context methods get_*_vector() should support template arguments for safe downcasting.  Like System::EvalVectorInput.
System::GetNumericParameter, which has downcasting support, should not exist... (the parameters can be evaluated purely from the context) and the context version should have the downcasting.

RussTedrake, 2017-09-04:
Note that even if the above are resolved, System::EvalVectorInput stands out as the only one of these that belongs to System (instead of Context).  I was taking for granted that you absolutely do need the System to (easily) evaluate the input... to populate the inputs to Context::EvalVectorInput.  But I'm not sure there is any fundamental reason why you shouldn't be able to evaluate an input directly on the Context (no System required).

sherm1, 2017-09-04:
A System is required whenever something has to be computed rather than just retrieved from the Context. That's an important architectural feature because the Context is intended to be an independent object not tied to any particular System object. In practice that means there are no pointers from a Context to the system, and that includes function pointers. The upside is that a Context can be (de)serialized, transmogrified, etc. without dragging a System along, and then used with any appropriate System object.

jwnimmer-tri, 2017-09-06:
I support this idea, but I'm going to de-assign myself until I'm tasked onto it.

jwnimmer-tri, 2024-03-16:
Closing as "not planned" -- our current architecture direction is to eschew subclassing of BasicVector, so the downcasting stuff will just evaporate someday anyway.

closed 2024-03-16 by jwnimmer-tri