RobotLocomotion/drake #6969: Consider returning Eigen::Ref<Compatible> rather than Block<Exact> or VectorBlock<Exact>

EricCousineau-TRI, 2017-09-05:
Per f2f with @amcastro-tri, it seems that using `Eigen::Ref<Compatible>` (where `Compatible` is a compatible matrix type) could possibly used as a replacement for returning a `Block<Exact>` or `VectorBlock<Exact>` (where `Exact` is the exact type of the upstream data object).

The rationale is that `Ref<PlainObjectType>` extends from `RefBase<PlainObjectType>`, which extends from `MapBase<PlainObjectType>`, and (I believe) it will store the structure of block, rather than a reference to a block itself.

This means that returning `Ref<>`s constructed from temporary / r-value `Block<>`s should not cause any memory leaks, e.g.:
```c++
Ref<VectorXd> SegmentOf(VectorXd& x) {
  return x.segment(0, 1);
}
```

The benefit is a little more encapsulation, such as being able to mix static- and dynamic-sized matrix references in the implementation, and possibly enabling stuff like passing references to `Eigen::Map<>` for other implementations (like something mapping a point cloud from VTK or PCL).

Here are some brief checks with Drake's profiling setup (memcheck, ASan, MSan, etc):
https://github.com/EricCousineau-TRI/drake/commit/09dd681260bdc76dd1f167707eedbc1688069bdf#diff-517a40a5aece9054c9b303aa8d88efbaR9

(TSan seems to puke a bit, but I don't think it pertains to memory issues specific to this.)

amcastro-tri, 2017-09-05:
So `segment()` creates (I believe) a `VectroBlock<VectorXd>` temporary object which now gets referenced by the newly created `Ref` object. I am actually impressed that the compiler doesn't puke there! Thoughts?

EricCousineau-TRI, 2017-09-05:
It doesn't puke because `Ref<>` will inherit the data pointer and structure defined by `VectorBlock<>`, and it does not need a reference to the object directly (which would require dirty-nasty template type erasure stuff) past its lifetime.

sherm1, 2017-09-05:
Ah, I didn't realize that `Eigen::Ref` derives from Map -- I thought it was just a pointer. Now I see it has the pointer, dimensions, and stride info. Thanks, @EricCousineau-TRI!

EricCousineau-TRI, 2017-09-26:
Gael's response was to use `Ref<>` with caution as it was intended for arguments, not returned values:
`Ref<Derived>` (where `Derived` is mutable) will ensure that it always references the original data, but `Ref<const Derived>` is a specialization that permits an implicit temporary copy to be stored if the incoming type does not have compatible continuity (stride, transpose, etc.) - following C++'s convention for implicit copies for `const T&` in C++.

The two main drawbacks are (a) performance and most importantly that (b) the reference to the original data is severed.

I posted a [workaround](https://forum.kde.org/viewtopic.php?f=74&t=141703#p381604) following the discussion in the bug mentioned (effectively re-implementing `Ref<>` without the implicit-copy specialization), and will see what feedback he might have if he has time.

EricCousineau-TRI, 2018-11-12:
Moving discussion from @edrumwri's [comment](https://reviewable.io/reviews/robotlocomotion/drake/9977#-LQuKdh1Erla_LohcVaU): `Eigen::Ref<>` permits calling `.resize()` at compile time, but will fail at runtime.

I still disprefer `VectorBlock<>` because it's too sensitive to the underlying type, and `Ref<>` permits this and type erasure (so that the underlying implementation can change); compile time checks are nice, but I don't think they should be the only criteria for usage.

sherm1, 2018-11-12:
A couple of questions on this:
- is there a problem that this change would solve for us?
- given that the main Eigen developer says that Ref is not for return arguments (see upthread), is it worth going against that advice?

jwnimmer-tri, 2020-12-09:
Given the two years lack of response to fundamental objections to this proposal, I'll say we should close this.

closed 2020-12-09 by jwnimmer-tri