RobotLocomotion/drake #6971: Add syntactic sugar for logically composing gtest AssertionResult's

EricCousineau-TRI, 2017-09-05:
When writing a few testing utilities, it can sometimes be a tad odd to logically AND or OR `AssertionResult`s, such as in `CompareTransformWithoutAxisSign`:
https://github.com/RobotLocomotion/drake/blob/f1d2dfe11cb88848427981f83b7d4181d5e0de84/drake/perception/estimators/dev/test/test_util.cc#L150

It'd be nice to be able to write something such as:
```c++
AssertionResult check_translation =
    CompareMatrices(X_expected.translation(), X_actual.translation(), 
                    tolerance)
    << "Translation";
AssertionResult check_rotation =
    CompareRotMatWithoutAxisSign(X_expected.rotation(), X_actual.rotation(),
                                 tolerance)
    << "Rotation";
return check_translation && check_rotation;
```

A small prototype:
https://github.com/EricCousineau-TRI/repro/blob/89acb4fb73caeec00fe2813f930ab7044cab53bc/cpp/gtest_op.cc

jwnimmer-tri, 2017-09-05:
Alternatively, my impression is that the "matcher" APIs are the preferred gtest way to denote and compose test predicates, e.g., https://github.com/google/googletest/blob/master/googlemock/docs/CheatSheet.md#floating-point-matchers and https://github.com/google/googletest/blob/master/googlemock/docs/CheatSheet.md#composite-matchers.  Manually dealing with `AssertionResult` by hand I think is dis-preferred.  However, we don't have very many examples of matchers yet in the tree.  I think it would be a fruitful area of infrastructure work, if someone was so inclined.

EricCousineau-TRI, 2017-09-06:
Cool, will try that out. Thanks!

jwnimmer-tri, 2020-05-28:
@EricCousineau-TRI could I convince you to close this as "not a current pain point"?  It doesn't seem likely to be actively worked anytime soon.

EricCousineau-TRI, 2020-05-28:
Yup!

EricCousineau-TRI, 2020-05-28:
(well, it is still kind of a pain point, but not a huge pain point)

closed 2020-05-28 by EricCousineau-TRI